gphoto/libgphoto2 #166: Libtool libltdl is not thread safe

kadler15, 2017-05-05:
I've been doing some work with libgphoto2 accessed from Node.js through a native add-on. I started seeing intermittent segfaults, with stack traces where multiple libuv threads were always centered around some calls to libltdl, usually during abilities list or port info list loading / detection.

E.g.:

```
Program terminated with signal SIGSEGV, Segmentation fault.
__strcmp_sse2_unaligned () at ../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S:29
29	../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S: No such file or directory.
(gdb) bt
#0  __strcmp_sse2_unaligned () at ../sysdeps/x86_64/multiarch/strcmp-sse2-unaligned.S:29
#1  0x00007ffff432d10f in ?? () from /usr/lib/x86_64-linux-gnu/libltdl.so.7
#2  0x00007ffff432d504 in ?? () from /usr/lib/x86_64-linux-gnu/libltdl.so.7
#3  0x00007ffff432f5c6 in ?? () from /usr/lib/x86_64-linux-gnu/libltdl.so.7
#4  0x00007ffff432f89d in lt_dlopenadvise () from /usr/lib/x86_64-linux-gnu/libltdl.so.7
#5  0x00007ffff432f9c0 in lt_dlopenext () from /usr/lib/x86_64-linux-gnu/libltdl.so.7
#6  0x00007ffff4744bf6 in gp_abilities_list_load_dir (list=list@entry=0x1e8d4b0, dir=<optimized out>, context=context@entry=0x1ef7da0) at gphoto2-abilities-list.c:210
#7  0x00007ffff4744ec9 in gp_abilities_list_load (list=0x1e8d4b0, context=0x1ef7da0) at gphoto2-abilities-list.c:308

```

I found a relevant piece of info in the ltdl docs:

[Thread safety in libltdl](https://www.gnu.org/software/libtool/manual/html_node/Thread-Safety-in-libltdl.html#Thread-Safety-in-libltdl)

> Libltdl provides a wrapper around whatever dynamic run-time object loading mechanisms are provided by the host system, many of which are themselves not thread safe. Consequently libltdl cannot itself be consistently thread safe.

> If you wish to use libltdl in a multithreaded environment, then you must mutex lock around libltdl calls, since they may in turn be calling non-thread-safe system calls on some target hosts.

To test the hypothesis that libltdl was causing the segfaults, I hacked in global locking around key libltdl calls. With the locking in place, the segfaults are no longer occurring. It's difficult to prove with total certainty that the issue is addressed, but my confidence is high.

I'm not suggesting this be merged as is. The code probably doesn't meet the project styling and structure standards, and may hinder portability. I also know there is no guarantee that multithreading with libgphoto2 is supported or tested. But I wanted to bring this issue to your attention, anyway.

https://github.com/kadler15/libgphoto2/commit/9ec3313a20b12cd11e100fb35f9b958725415b83

RReverser, 2022-09-24:
Running into this in Rust bindings as well; would be good to merge the proposed fixes if they work.

RReverser, 2022-10-07:
Sorry, but... ping @msmeissn.

I've seen some threads on sourceforge where you mentioned that gphoto2 should be MT-safe (as long as you access different devices), so I believe this is the only issue preventing it from being actually multithreading-friendly?

msmeissn, 2022-10-08:
well, i merged the above referenced pthread locking into a mutex branch for review.
I am not sure abiout how adding pthread dependency will break stuff or not.

RReverser, 2022-10-08:
> I am not sure abiout how adding pthread dependency will break stuff or not.

Given that libusb already depends on pthread, and most users of gphoto2, I suspect, use it with libusb support, it shouldn't make much difference?

ndim, 2022-10-08:
So we can wrap all `lt_*()` calls between `pthread_mutex_lock(&gpi_libltdl_mutex);` and `pthread_mutex_lock(&gpi_libltdl_mutex);`. That is the easy part (see e.g. https://github.com/gphoto/libgphoto2/pull/848).

However, what happens when an application linking against libgphoto2 uses `lt_*()` on its own or one of its other libraries uses `lt_*()`?

RReverser, 2022-10-08:
> However, what happens when an application linking against libgphoto2 uses `lt_*()` on its own or one of its other libraries uses `lt_*()`?

Yeah that's a good question and why reasoning at https://www.gnu.org/software/libtool/manual/html_node/Thread-Safety-in-libltdl.html doesn't make much sense - however imperfect that mutex locking API was, it belongs to libltdl.

That said, I don't think many apps use libltdl, and doing locking at least in libgphoto2 would be much better than the current status quo anyway.

ndim, 2022-10-08:
> > I am not sure abiout how adding pthread dependency will break stuff or not.
> 
> Given that libusb already unconditionally depends on pthread, and most users of gphoto2, I suspect, use it with libusb support, it shouldn't make much difference?

On Linux, correct. However, there are operating systems which have their own implementations of the libusb API (FreeBSD and Windows come to mind). Those other implementations of the libusb API might or might not use pthread.

RReverser, 2022-10-08:
Even if they don't, if the concern is adding dependency on pthread in principle (as in, that there are users who don't have it), then libgphoto2 already unconditionally checks for it during the configure stage on all platforms: https://github.com/gphoto/libgphoto2/blob/e47c13fd8ebc24932ec58078369fe325beb3ea64/libgphoto2_port/configure.ac#L358

RReverser, 2022-10-08:
> Yeah that's a good question and why reasoning at [gnu.org/software/libtool/manual/html_node/Thread-Safety-in-libltdl.html](https://www.gnu.org/software/libtool/manual/html_node/Thread-Safety-in-libltdl.html) doesn't make much sense - however imperfect that mutex locking API was, it belongs to libltdl.
> 
> That said, I don't think many apps use libltdl, and doing locking at least in libgphoto2 would be much better than the current status quo anyway.

Thinking about this a bit more, perhaps the right solution here would be to link libltdl statically instead of dynamically? Then it could coexist with another instance of libltdl in the app itself without any thread-safety issues, as each instance would be isolated from each other.

ndim, 2022-10-09:
This far, we have avoided using the `LTDL_INIT()` configure.ac macro because that insists on copying the libltdl source code into out source tree and distributing the libltdl source code with the libgphoto2 dist tarball, and we wanted to avoid shipping foreign source code. After all, that means we need to track that foreign code and maybe even put out our own release just because they fixed a bug in their release, and now suddenly all of our older releases should both be kept as they are because "do not change history" and be changed to the fixed foreign code.

Just statically linking libgphoto2 against `libtool.a` cannot be relied on as static libs are not routinely part of libltdl distro packages (again, using Fedora as an example).

When installing `libtool` with `make install`, actually the whole libltdl source code is installed so that `libtoolize` can copy it when `LTDL_INIT()` is present.

So if there is a libltdl-devel package installed which contains the `libltdl.so` for us to link against, we can probably also assume the presence of the libltdl sources in `/usr/share/libtool` and could use those to compile and statically link into our library.

If we go the "official way" with the m4 macros libtool provides us with and accept the drawback of shipping foreign source code, we would do something like the following:

```configure.ac
LT_CONFIG_LTDL_DIR([libltdl])
LTDL_INIT([convenience])
```

but that still leaves us with the problem that we would now require the user to call `configure` with `--with-included-ltdl` in order to actually use our convencience library (i.e. statically compile libltdl into libgphoto2.so) - otherwise still the system wide libltdl.so is linked against dynamically.

It appears to be possible to change the default value of `--with-included-ltdl`, though (with a bit of a hack):

```configure.ac
LT_CONFIG_LTDL_DIR([libltdl])
if test "x${with_included_ltdl-unset}" = xunset; then
  with_included_ltdl=yes
fi
LTDL_INIT([convenience recursive])
```

However, the libgphoto2 code still needs to guard all of its `lt_*()` function calls against concurrent execution, though.

It still would be easier if libltdl just were thread safe.

RReverser, 2022-10-09:
> However, the libgphoto2 code still needs to guard all of its `lt_*()` function calls against concurrent execution, though.

Yeah that's still necessary either way, and will be resolved by #848. Those extra steps are only for the (rare) case of app depending on libltdl too.

> It still would be easier if libltdl just were thread safe.

I agree that would be ideal, but... it just isn't, so we have to work around that :(

---

Side-stepping the issues with static linking, I actually remembered about `dlmopen`. I never used it in the past, but I wonder if that could help here - my understanding from the docs is that, with `LM_ID_NEWLM`, it would provide exactly the kind of isolated copy of a dynamic library that we want?

Of course, it would be slightly more awkward to invoke than implicit dynamic linking, but shouldn't be too bad, and it would allow to avoid the issues you mentioned with having to keep up with libltdl developments.

RReverser, 2022-10-09:
OTOH, I really think that using libltdl is not that common - in fact, libgphoto2 was the first one I saw using it - so perhaps all of this is a bit premature and for now #848 would be enough.

closed 2022-11-05 by msmeissn