gphoto/libgphoto2 #168: Reset camera status after preview capture

irieger, 2017-05-16:
Hello,

I'm working on a small tool to control DSLR via a small Qt application. The basics are working well with different (Canon) cameras. Mostly I'm working and testing with the 5D mk III and the 5Ds.

Besides the basics like setting image format, shutter speed, aperture etc. and capturing, which are all working, I try to get a live view. The code is oriented at the workings of capture-movie in gphoto and just pushes frames from capture_preview into a fifo. That works at first start, but after I start my preview once and stop it then, the camera blocks ("busy") and I can't control anything (no capture, no new preview etc. Only the basic setting of config parameters works) either with libgphoto or on the body unless I power cycle it or at least unplug the USB.

So any idea why the camera blocks after this? Also the mirror will flip away but isn't switching back in (which kind of makes sense as it would have to switch after any call of capture_preview in the loop ...).

P.S. Is there a way to get a crop-in preview for focus check? Couldn't find anything with my searches.

mickeprag, 2017-11-09:
I have experienced the same as you. My workaround is to capture a real image after I am done with preview. This resets the mirror. It works with Nikon D5000.
Not pretty but works in my case.

galazzo, 2017-11-11:
I experience the same problem with Canon 5D mark IV and it's really annoying.
I see this bug is open since 16 May, any chance to have a fix?

TermanEmil, 2019-07-15:
I have the same problem as well

elijahparker, 2019-07-15:
With most cameras you can reset the liveview mode and bring the mirror back
down by setting viewfinder=0

TermanEmil, 2019-07-16:
I'm using Nikon Z6. I managed to reproduce the problem both on a mac and on centOS.
What I'm trying to do is similar to this:
~~~bash
while :
do
	gphoto2 --capture-preview --stdout > /dev/null;
done
~~~

Sometimes (quite often), if I stop this process and then run it again, the access to the camera is blocked.
Even if I run `gphoto2 --set-config-value viewfinder=0;`, the connection is still blocked, meaning that this command doesn't end.

Do you have any more hints?

After I got into a situation where the camera was blocked, I run this:
~~~
gphoto2 --debug --set-config-value viewfinder=0;
0.000081 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000190 main                        (2): gphoto2 2.5.23
0.000232 main                        (2): gphoto2 has been compiled with the following options:
0.000257 main                        (2):  + gcc (C compiler used)
0.000295 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000343 main                        (2):  + exif (for displaying EXIF information)
0.000379 main                        (2):  + no cdk (for accessing configuration options)
0.000422 main                        (2):  + no aa (for displaying live previews)
0.000453 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000486 main                        (2):  + readline (for easy navigation in the shell)
0.000528 main                        (2): libgphoto2 2.5.23
0.000562 main                        (2): libgphoto2 has been compiled with the following options:
0.000587 main                        (2):  + all camlibs
0.000616 main                        (2):  + gcc (C compiler used)
0.000649 main                        (2):  + ltdl (for portable loading of camlibs)
0.000672 main                        (2):  + EXIF (for special handling of EXIF files)
0.000700 main                        (2): libgphoto2_port 0.12.0
0.000734 main                        (2): libgphoto2_port has been compiled with the following options:
0.000756 main                        (2):  + iolibs: disk ptpip serial usb usbdiskdirect usbscsi
0.000785 main                        (2):  + gcc (C compiler used)
0.000807 main                        (2):  + ltdl (for portable loading of iolibs)
0.000838 main                        (2):  + USB (libusb0, for USB cameras)
0.000860 main                        (2):  + serial (for serial cameras)
0.000882 main                        (2):  + no resmgr (serial port access and locking)
0.000913 main                        (2):  + no ttylock (serial port locking)
0.000936 main                        (2):  + no lockdev (serial port locking)
0.000963 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000985 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001013 main                        (2): invoked with following arguments:
0.001036 main                        (2):   --debug
0.001057 main                        (2):   --set-config-value
0.001079 main                        (2):   viewfinder=0
0.001139 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.001341 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.001827 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.001913 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.0'...
0.002281 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.003033 gp_port_library_list        (2): found mtab fsname rootfs
0.003170 gp_port_library_list        (2): found mtab fsname overlay
0.003234 gp_port_library_list        (2): found mtab fsname proc
0.003280 gp_port_library_list        (2): found mtab fsname tmpfs
0.003365 gp_port_library_list        (2): found mtab fsname devpts
0.003385 gp_port_library_list        (2): found mtab fsname sysfs
0.003405 gp_port_library_list        (2): found mtab fsname tmpfs
0.003446 gp_port_library_list        (2): found mtab fsname cgroup
0.003459 gp_port_library_list        (2): found mtab fsname cgroup
0.003470 gp_port_library_list        (2): found mtab fsname cgroup
0.003483 gp_port_library_list        (2): found mtab fsname cgroup
0.003498 gp_port_library_list        (2): found mtab fsname cgroup
0.003550 gp_port_library_list        (2): found mtab fsname cgroup
0.003583 gp_port_library_list        (2): found mtab fsname cgroup
0.003604 gp_port_library_list        (2): found mtab fsname cgroup
0.003621 gp_port_library_list        (2): found mtab fsname cgroup
[... 197 lines cut ...]
96.659313 camera_init [library.c:8474](0): 'ptp_opensession (params, sessionid)' failed: PTP General Error (0x2002)
96.659324 camera_init [library.c:8502](0): 'ret' failed: 'PTP General Error' (0x2002)
96.659337 gp_context_error            (0): PTP General Error

*** Error ***              
PTP General Error
96.659381 gp_port_close               (2): Closing port...
96.659667 gp_context_error            (0): An error occurred in the io-library ('Unspecified error'): No error description available

*** Error ***              
An error occurred in the io-library ('Unspecified error'): No error description available
*** Error (-1: 'Unspecified error') ***       

[... cut ...]
96.660469 gp_filesystem_reset         (2): resetting filesystem
96.660482 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
96.660490 gp_filesystem_lru_clear     (2): fscache LRU list already empty
96.660498 delete_all_folders          (2): Internally deleting all folders from '/'...
96.660508 lookup_folder               (2): Lookup folder '/'...
96.660522 lookup_folder               (2): Found! / is 0x5568915b1120
96.660534 recurse_delete_folder       (2): Recurse delete folder 0x5568915b1120//
~~~

msmeissn, 2019-07-24:
The Nikon Z6 issue is seperate from the Canon part.

Here you confuse the PTP communication where it cannot be picked up again. You should not abort the process, but perhaps use a trigger file to stop the loop or similar so it cleanly exits.

msmeissn, 2019-07-24:
For the Canon users, did gphoto2 --set-config viewfinder=0   work for you?

andreakappa, 2020-01-24:
I'm working with an EOS 5Ds, and I have this problem too by controlling it from python. 
I've managed to put an handler with `atexit`that runs something like this:

`    def close(self):
        gp.check_result(gp.gp_camera_exit(self.camera))`

after some time  I start getting timeout errors, and even a `gphoto2 --summary` fails.