gphoto/libgphoto2 #169: Cannot compile due to autoconf error

daparker, 2017-05-17:
I just downloaded the zip from GitHub because I'd like to get the fix for the iOS 10.2/10.3 PTP issue.  However, when I run `autoreconf --install --symlink` as indicated in Step 0 in the INSTALL file, it copies a bunch of files and then fails with this error:

```
configure.ac:75: error: possibly undefined macro: AC_DISABLE_STATIC
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:82: error: possibly undefined macro: AC_LIBTOOL_WIN32_DLL
configure.ac:83: error: possibly undefined macro: AC_LIBTOOL_DLOPEN
configure.ac:84: error: possibly undefined macro: AC_PROG_LIBTOOL
autoreconf2.50: /usr/bin/autoconf failed with exit status: 1
```

Unfortunately, I do not know enough about autoconf or M4 to troubleshoot this too deeply.  OS is Debian 8 (64-bit), and I'm using automake and autoconf installed from Debian packages:

```
$ automake --version
automake (GNU automake) 1.14.1

$ autoconf --version
autoconf (GNU Autoconf) 2.69
```

Any help would be greatly appreciated.  Thanks!

msmeissn, 2017-05-18:
(the ios fix is also in the 2.5.13 release if you just want the tarball)

it seems libtool macros are missing. is the libtool / libtool-devel package installed?

daparker, 2017-05-20:
Thanks.  It turns out I was missing the `libtool` package, and installing it got me past the `autoconf` errors.  However, I ran into a few more problems.

First, after the `autoreconf` step succeeded, the `make` step failed with this:

```
make[6]: Entering directory '/home/dparker/libgphoto2-master/libgphoto2_port/po'
: --default-domain=libgphoto2_port-12 --directory=.. \
  --add-comments=TRANSLATORS: --keyword=_ --keyword=N_ \
  --files-from=./POTFILES.in \
  --copyright-holder='Lutz Mueller and others' \
  --msgid-bugs-address='gphoto-devel@lists.sourceforge.net'
test ! -f libgphoto2_port-12.po || { \
  if test -f ./libgphoto2_port-12.pot; then \
    sed -f remove-potcdate.sed < ./libgphoto2_port-12.pot > libgphoto2_port-12.1po && \
    sed -f remove-potcdate.sed < libgphoto2_port-12.po > libgphoto2_port-12.2po && \
    if cmp libgphoto2_port-12.1po libgphoto2_port-12.2po >/dev/null 2>&1; then \
      rm -f libgphoto2_port-12.1po libgphoto2_port-12.2po libgphoto2_port-12.po; \
    else \
      rm -f libgphoto2_port-12.1po libgphoto2_port-12.2po ./libgphoto2_port-12.pot && \
      mv libgphoto2_port-12.po ./libgphoto2_port-12.pot; \
    fi; \
  else \
    mv libgphoto2_port-12.po ./libgphoto2_port-12.pot; \
  fi; \
}
make[6]: Leaving directory '/home/dparker/libgphoto2-master/libgphoto2_port/po'
: --update cs.po libgphoto2_port-12.pot
rm -f cs.gmo && : -c --statistics -o cs.gmo cs.po
mv: cannot stat ‘t-cs.gmo’: No such file or directory
Makefile:117: recipe for target 'cs.gmo' failed
make[5]: *** [cs.gmo] Error 1
make[5]: Leaving directory '/home/dparker/libgphoto2-master/libgphoto2_port/po'
Makefile:141: recipe for target 'stamp-po' failed
make[4]: *** [stamp-po] Error 2
make[4]: Leaving directory '/home/dparker/libgphoto2-master/libgphoto2_port/po'
Makefile:1171: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/dparker/libgphoto2-master/libgphoto2_port'
Makefile:670: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/dparker/libgphoto2-master/libgphoto2_port'
Makefile:678: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/dparker/libgphoto2-master'
Makefile:500: recipe for target 'all' failed
make: *** [all] Error 2
```

There is indeed no file called t-cs.gmo anywhere in the libgphoto2-master folder.

I compiled and installed from the tarball, and was eventually able to get the photos off the iPhone using the most recent version of gphoto2 with libgphoto2 2.5.13.

Thanks!

msmeissn, 2017-05-20:
these are now missing gettext utiltities, especially msgfmt    install gettext and friends