gphoto/libgphoto2 #170: Fujifilm x-t20 not working

whiteud, 2017-05-28:
gphoto2 --capture-image
                                                                               
*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.

[log.txt](https://github.com/gphoto/libgphoto2/files/1034534/log.txt)

msmeissn, 2017-06-02:
The Fuji XT-2 started having remote capture after a firmware update.

(And with current GIT currently, to be in  2.5.14 or later)

whiteud, 2017-06-02:
Fuji x-t20 is other camera, like x-t10, not x-t2

msmeissn, 2017-06-15:
hmm, some google research does not find indications of this happening.

so its probably not supported by the current XT-20 camera firmware.

closed 2017-06-15 by msmeissn