gphoto/libgphoto2 #173: Fuji XT-2 detected as Nikon DSC D750

22u2, 2017-06-08:
I have just compiled libgphoto2 and gphoto 2.5.14 to check if I can browse into my XT-2 (Firmware 2.10) but it does not work.

gphoto2 --auto-detect
Nikon DSC D750                 usb:001,001    

[mylog.txt](https://github.com/gphoto/libgphoto2/files/1062406/mylog.txt)

msmeissn, 2017-06-09:
This means you built without libusb support and it fell back to our internal "virtual" testcamera.

install libusb 1 development headers/libraries and reconfigure / remake

22u2, 2017-06-09:
Thanks it works.
Is it reasonable to make it mandatory during the configure step?

closed 2017-06-09 by 22u2

msmeissn, 2017-06-15:
possible, but i can envision scenarios where people might want to build without.

perhaps the other way round, enable the virtual driver only when expclitily enabled