gphoto/libgphoto2 #174: undefined referance to gpi_vnsprintf in libgphoto2-2.5.14 after master(20170611) failed.

scsijon, 2017-06-11:
Firstly I downloaded and tried to build the master but as it had no configure it failed miserably.

I'm building in T2, x86-64 and using gcc v7.1, glibc v2.25 and glib v2.53.1.

Reverted to 2.5.14 and some 2468-71 lines in, it came across a problem for you I think :-

../libgphoto2/.libs/libgphoto2.so: undefined referance to 'gpi_vsnprintf'
collect2: error: ld returned 1 exit status
make[3]: *** [makefile:873: test-gphoto2] Error 1
make[3]: Leaving directory 'TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests'

Can you assist please or is there a way of turning the tests off (reluctantly) as i'm now down to the last few dozen problems.

If you want the whole errorlog let me know, it's 371k though.

thanks

msmeissn, 2017-06-11:
This is weird. The function is there and has been there for a while. 

This would mean that libtool did not want to link libgphoto2_port 

Just out of a gut feeling, install libexif development headers and libraries :/

msmeissn, 2017-06-11:
(Can you attach the full buildlog, or at least some lines before the error)

scsijon, 2017-06-11:
libexif is already installed.

it won't allow me to attach the full error file so lines between 2453-2476 are below. How many lines can it handle at a time, maybe I need to split it if you want all of them?

Making all in tests
make[2]: Entering directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests'
Making all in ddb
make[3]: Entering directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests/ddb'
make  all-am
make[4]: Entering directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests/ddb'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests/ddb'
make[3]: Leaving directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests/ddb'
make[3]: Entering directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests'
depbase=`echo test-gphoto2.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I. -I..  -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/usr/share/locale\" -DCAMLIBS=\"/usr/lib64/libgphoto2/2.5.14\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2   -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -MT test-gphoto2.o -MD -MP -MF $depbase.Tpo -c -o test-gphoto2.o test-gphoto2.c &&\
mv -f $depbase.Tpo $depbase.Po
/bin/sh ../libtool  --tag=CC   --mode=link gcc  -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes   -o test-gphoto2 test-gphoto2.o ../libgphoto2/libgphoto2.la ../libgphoto2_port/libgphoto2_port/libgphoto2_port.la -lltdl -lexif  
libtool: link: gcc -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -o .libs/test-gphoto2 test-gphoto2.o  ../libgphoto2/.libs/libgphoto2.so /TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/libgphoto2_port/libgphoto2_port/.libs/libgphoto2_port.so ../libgphoto2_port/libgphoto2_port/.libs/libgphoto2_port.so -lpthread /usr/lib64/libltdl.so -ldl /usr/lib64/libexif.so -lm
../libgphoto2/.libs/libgphoto2.so: undefined reference to `gpi_vsnprintf'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:873: test-gphoto2] Error 1
make[3]: Leaving directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests'
make[2]: *** [Makefile:985: all-recursive] Error 1
make[2]: Leaving directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/tests'
make[1]: *** [Makefile:689: all-recursive] Error 1
make[1]: Leaving directory '/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14'
make: *** [Makefile:512: all] Error 2

msmeissn, 2017-06-11:
so libtool has decided it does not need libgphoto2_port even though being told so :/
 
 /bin/sh ../libtool --tag=CC --mode=link gcc -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -o test-gphoto2 test-gphoto2.o ../libgphoto2/libgphoto2.la ../libgphoto2_port/libgphoto2_port/libgphoto2_port.la -lltdl -lexif

 libtool: link: gcc -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -o .libs/test-gphoto2 test-gphoto2.o ../libgphoto2/.libs/libgphoto2.so /TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/libgphoto2_port/libgphoto2_port/.libs/libgphoto2_port.so ../libgphoto2_port/libgphoto2_port/.libs/libgphoto2_port.so -lpthread /usr/lib64/libltdl.so -ldl /usr/lib64/libexif.so -lm

the last line should have -lgphoto2_port

can you attach libgphoto2_port/libgphoto2_port/libgphoto2_port.la and libgphoto2/libgphoto2.la ?

scsijon, 2017-06-11:
Ok here they are I think packaged up of course

[Uploading libgphoto2-port.tar.gz…]()
[Uploading libgphoto2-la.tar.gz…]()

will try full errorlog if this one works as think I have found the upload problem!

scsijon, 2017-06-11:
[errorlog-libgphoto-2514.tar.gz](https://github.com/gphoto/libgphoto2/files/1066363/errorlog-libgphoto-2514.tar.gz)
yup, firewall at isp had github partly blocked "due to unexpected high file traffic". It's fixed, at least for now!

msmeissn, 2017-06-12:
the last upload was done, but the first two uploads were somehow not

scsijon, 2017-06-12:
ok we will try again

[libgphoto2-la.tar.gz](https://github.com/gphoto/libgphoto2/files/1067858/libgphoto2-la.tar.gz)
[libgphoto2-port.tar.gz](https://github.com/gphoto/libgphoto2/files/1067859/libgphoto2-port.tar.gz)

msmeissn, 2017-06-15:
hmm, so far nothing visible... stupid libtool stuff :( 

can you also attach 

/TOOLCHAIN/src.libgphoto.racy6.20170611.154603.20693.localhost/libgphoto2-2.5.14/libgphoto2_port/libgphoto2_port/libgphoto2_port.la

scsijon, 2017-06-16:
just did a new complete (sort of) build cycle as 12f. Attaching all three files in a .gz.
[libgphoto2-2.5.14-c12f.tar.gz](https://github.com/gphoto/libgphoto2/files/1079677/libgphoto2-2.5.14-c12f.tar.gz)

2 questions if I may from reading the errorlog?

?Interesting in that i've got libgd (2.2.4) installed before it, but according to lines 222 / 223 and 660 it's not? There's /usr/lib64/libgd.so and /usr/include/gd.h and the rest, is it looking for one of these or something totally different?

?And as the second question, ?Is it using libusb or libusb-compat when it refers to libusb? I have both installed and wonder if there is cross interfearance although the filenames are different, such as libusb-1.0.a in libusb, and libusb.a in libusb-compat, both are used by different packages so both are needed. However libusb is v1.0.x and libusb-compat is 0.1.5 but libusb-compat has /usr/include/usb.h while libusb has libusb.h (have a look around lines 578 to 594 in the error log too see what I mean).

I'm down to six direct problems in this build cycle (started with over 180) so would love to close it out,  although I can easily roll it over to the next cycle if you ask and as I have done with some others because of complexity, time constraints or the maintainers needing to get their heads around gcc7 (no java, tighter code, etc). The next cycle (strange that it's no 13) has all the nasty build problems in it!

regards and thanks for your help and time spent to date
scsijon

scsijon, 2017-06-17:
HI, do you remember this?
https://sourceforge.net/p/gphoto/bugs/1020/feed.rss

I only came across it this morning.

As i'm now down to 3 direct problems in this build cycle I thought i'd look around... while it was building. OK, truth tells, I was bored with both my two builder workstations busy compiling!

Anyway, after a lot of playing around working out how to disable the autosetting in T2, not easy but I finally did it.
And 'yes' the build problem has disapeared and I have a completed build.
So it looks like the root cause of the problem hasn't disapeared. I know it failed in 2.5.10, 2.5.13 and 2.5.14 (not sure about earlier ones).
Do you want a log file that shows it building sucessfully to compare with the failed one?

I had also found on debian some patches that seemed to work, although I haven't yet checked to see if they were replacing like for like. Do you want them ?

OH, and it looks like I will be back to someone as gphotofs has just failed gphotofs.c:673:15 undeclaired identifier.

regards
scsijon

leleliu008, 2022-11-06:
`gpi_vsnprintf` function only available when `--disable-debug` switch is not given. if `--disable-debug` is given, this error will be triggered.

msmeissn, 2022-11-07:
thanks, this detail was missing... should be fixed now.

closed 2022-11-07 by msmeissn