gphoto/libgphoto2 #176: Trouble installing gphoto2 using pip. ERROR: can not execute: "pkg-config --modversion libgphoto2"

pavelzag, 2017-06-20:
Hi there!
I am attempting to install gphoto2 using pip for my Python project that runs on Raspberry Pi1 with DietPy.
This is the error I am receiving when attempting to install:

(timelapse_virt) root@DietPi:/home/dietpi/timelapse# pip install gphoto2
Collecting gphoto2
  Using cached gphoto2-1.7.0.tar.gz
    Complete output from command python setup.py egg_info:
    ERROR: can not execute: "pkg-config --modversion libgphoto2"
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-f9LKP0/gphoto2/setup.py", line 37, in <module>
        cmd, universal_newlines=True).split('.')
      File "/usr/lib/python2.7/subprocess.py", line 566, in check_output
        process = Popen(stdout=PIPE, *popenargs, **kwargs)
      File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
        errread, errwrite)
      File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
        raise child_exception
    OSError: [Errno 2] No such file or directory

When attempting to install through apt I succeed..
apg-get install gphoto2

Thanks for the assistance!
Pavel

msmeissn, 2017-06-22:
the error seems to imply that you need to install the pkg-config package. (and libgphoto2 development headers)

pavelzag, 2017-06-26:
Thanks a lot! This helped with the issue

closed 2017-06-26 by pavelzag

Mas9311, 2019-11-25:
Thanks @msmeissn 

I came across this error for the gphoto2 package when upgrading from 1.9.0 => 2.0.0
Did **not** fix the issue by _just_ installing the `libgphoto2-6` package (current at time of writing).
Had to install the `libgphoto2-dev` package to fix the issue:
$ `sudo apt install libgphoto2-dev`

msmeissn, 2019-12-04:
yes. the libgphoto2 development headers i mentioned are in the -dev package and are needed for building, correct