gphoto/libgphoto2 #179: Problem with reading serial.

mohacs, 2017-07-03:
Hello everyone, 

I am having problem related reading serial number from Canon devices. Mostly we are using EOS 1D MARK III, 1DX etc.

gphoto2 --summary returns serial number does not match with the label. actually it doesn't looks like it is correct one. Here is a sample "6e8945f39baa4aa695c099dac408b12b". Actual serial number on the label 576xxx.

Do I need a conversation?

Thanks in advance.
-mohac.

msmeissn, 2017-07-03:
can you check the "eosserialnumber" configuration value?

serialnumber is the PTP reported serialnubmer, eosserialnumber the other one.

mohacs, 2017-07-03:
Hi Martin,

When call --list-config these are the listed. Only serial number related config value is /main/status/serialnumber... Still tried --get-config=/main/status/eosserialnumber however it returns nothing than "Label: Camera Status Information".

Camera EOS 1D Mark III
FW: 3-1.2.3

Getting same result with 1DX as well.

/main/actions/bulb                                                             
/main/actions/uilock
/main/actions/syncdatetime
/main/actions/autofocusdrive
/main/actions/manualfocusdrive
/main/actions/eoszoom
/main/actions/eoszoomposition
/main/actions/eosviewfinder
/main/settings/datetime
/main/settings/output
/main/settings/evfmode
/main/settings/ownername
/main/settings/customfuncex
/main/settings/autopoweroff
/main/settings/capturetarget
/main/settings/capture
/main/status/serialnumber
/main/status/manufacturer
/main/status/cameramodel
/main/status/deviceversion
/main/status/vendorextension
/main/status/model
/main/status/ptpversion
/main/status/dpofversion
/main/status/availableshots
/main/imgsettings/imageformat
/main/imgsettings/imageformatsd
/main/imgsettings/imageformatcf
/main/imgsettings/imageformatexthd
/main/imgsettings/iso
/main/imgsettings/whitebalance
/main/imgsettings/colortemperature
/main/imgsettings/whitebalanceadjusta
/main/imgsettings/whitebalanceadjustb
/main/imgsettings/whitebalancexa
/main/imgsettings/whitebalancexb
/main/imgsettings/colorspace
/main/capturesettings/exposurecompensation
/main/capturesettings/focusmode
/main/capturesettings/autoexposuremode
/main/capturesettings/picturestyle
/main/capturesettings/aperture
/main/capturesettings/shutterspeed
/main/capturesettings/meteringmode
/main/capturesettings/bracketmode
/main/other/d402
/main/other/d407
/main/other/d406

Thank you so much,
-mohac.

msmeissn, 2017-07-03:
can you run 
gphoto2 --list-config --debug --debug-logfile=list-config.log
and attach list-config.log here?

mohacs, 2017-07-03:
Here it is as requested. 
Thank you,
-mohac.
[list-config.txt](https://github.com/gphoto/libgphoto2/files/1120134/list-config.txt)

mohacs, 2017-07-08:
Hi Marcus, 
Did you find a chance to have a look?
Thank you,
-mohac.

msmeissn, 2017-07-09:
hmm,. the download of list-config.txt did not work? can you reupload?

mohacs, 2017-07-09:
Hi Marcus,
Here it is.
[list-config.txt](https://github.com/gphoto/libgphoto2/files/1134089/list-config.txt)

Also uploaded to syncplicity.  https://my.syncplicity.com/share/avfuajunppw6xss/list-config

msmeissn, 2017-07-12:
its not visible in the trace. "576..." is not there as string and if it is 6 positions, also not as hex number.

We try to also request it from the camera, but the camera reacts with error:
0.594020 camera_prepare_canon_eos_capture [config.c:358](0): 'ptp_canon_eos_requestdevicepropvalue (params, PTP_DPC_CANON_EOS_SerialNumber)' failed: PTP General Error (0x2002)

I would say that the camera might not expose it.

mohacs, 2017-07-12:
Hi Marcus,
Thank you so much looking in to it. Then we can come to a conclusion Canon does not expose serial number because I have tried with EOS 1D, 1DX, 4D and 5D. 
Again thank you for your time. 
Cheers,
-mohac.

closed 2017-07-12 by mohacs