gphoto/libgphoto2 #181: Canon error 70 while calling --list-files, --get-file, and --delete-file

J0hannB, 2017-07-10:
Background:

I am working on an application where I automate the capturing of images from an external trigger and then copy those photos to a PC using gphoto2

Setup:

I am using a python script and the subprocess module to call gphoto2 commands. I repeatedly call --list-files to see if any new images have been taken, and when I find some, iterate through them calling --get-file and --delete-file for each.  My camera is a Canon 6D

Problem:

This works great for a while, but after many pictures have been copied over I inevitably see the infamous err 70 on the camera's LCD, and it refuses to respond to gphoto2 (or take pictures or anything else for that matter) until I power off, remove and reinsert the battery and SD card, and power on. 

I was originally using Jim Easterbrook's python-gphoto2 interface to copy the files and ran into the same issue. I tried taking hundreds of images and copying them all by calling --get-all-files directly from the terminal and it worked every time. Thinking it was a python-gphoto2 issue, I tried the subprocess.call() setup I am using now. 

I don't have a log file at the moment, but I can try to produce one soon. Any help would be appreciated!

J0hannB, 2017-07-10:
Here is the log file
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/1137095/my-logfile.txt)

J0hannB, 2017-07-11:
Here is the relevant code. I added the time.sleep(2) lines to make sure that the issue wasn't caused by reading/copying/deleting too quickly. But it did not solve the issue, and I eventually need the copying to happen as quickly as possible.

```
def watch_files_subprocess():
    while True:
        computer_files = list_computer_files()
        try:
            print "listing files"
            images = list()
            cameraIOLock.acquire()
            imageOutput = subprocess32.check_output("gphoto2 --debug --debug-logfile=logs.txt --list-files",
                                                    shell=True,
                                                    timeout=30)
            cameraIOLock.release()
            lines = imageOutput.split("\n")
            for i in range(0, len(lines)):
                line = lines[i]
                count = 0
                path = ""
                if line.startswith("There are "):
                    countStr = line.split("There are ")[1].split(" files in folder")[0]
                    count = int(countStr)
                    path = line.split("files in folder '")[1][:-2]
                elif line.startswith("There is 1"):
                    count = 1
                    path = line.split("file in folder '")[1][:-2]
                else:
                    continue

                for j in range(i+1, i+count):
                    line = lines[j]
                    for s in line.split(" "):
                        if s[-4:].lower() == ".jpg":
                            images.append(path + "/" + s)

                i = i + count

        except Exception as ex:
            template = "An exception of type {0} occurred. Arguments:\n{1!r}"
            message = template.format(type(ex).__name__, ex.args)
            print message
            print "Error listing files. Trying again"
            continue

        time.sleep(2)

        if images:

            for path in images:
                folder, name = os.path.split(path)
                dest = os.path.join(imageDir, name)

                if dest in computer_files:
                    # continue
                    tries = 0
                    newDest = dest
                    while newDest in computer_files:
                        tries+=1
                        newDest = "%s(%d)" % (dest, tries)
                    os.rename(dest, newDest)

                print "Copying %s to %s" % (path, dest)
                try:
                    cameraIOLock.acquire()
                    subprocess32.call("gphoto2 --debug --debug-logfile=logs --get-file " + path,
                                      cwd=imageDir+"/",
                                      shell=True,
                                      timeout=10)

                    time.sleep(2)

                    subprocess32.call("gphoto2 --debug --debug-logfile=logs.txt --delete-file " + path,
                                      cwd=imageDir+"/",
                                      shell=True,
                                      timeout=5)

                    time.sleep(2)

                    cameraIOLock.release()
                except Exception as ex:
                    template = "An exception of type {0} occurred. Arguments:\n{1!r}"
                    message = template.format(type(ex).__name__, ex.args)
                    print message
                    print "Error copying files. Trying again"
                    continue
```

J0hannB, 2017-07-11:
Just realizing that I probably should have submitted this issue to the gphoto2 repo rather than libgphoto2...

msmeissn, 2017-07-11:
(it is fine here, if there is an actual bug it is probably in the library)

msmeissn, 2017-07-11:
If this is continously running, you can potentially also use a single
gphoto2 --wait-event-and-download=10000s 
call (just extend the wait time)

The logfile looks successful.

I have not heard of this yet, but this is probably not a direct libgpoto2 issue :/

J0hannB, 2017-07-11:
Marcus, thanks for the quick reply!

Well, if there were ever a good lesson as to why you should read _all_ of the documentation, this is it. I think `--wait-event-and-download` is exactly what I need. I don't know why I never noticed it before. I'm doing some long term tests with it now to make sure I don't see err 70 again, but I haven't had any issues so far.

J0hannB, 2017-07-11:
I was able to download 9641 pictures while the camera was being externally triggered over the course of about three hours without a problem until I ran into:
```
                                                                               
*** Error ***              
PTP I/O Error
*** Error (-7: 'I/O problem') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --wait-event-and-download

Please make sure there is sufficient quoting around the arguments.
```

J0hannB, 2017-07-11:
I  just tried running the `--wait-event-and-download` command from within python, and now I see the camera lock up and say "busy" after about 50 images, but the output form gphoto2 never indicates an error (keeps printing out camera events).

New code :
```
    def watch_files(self):
        with self.cameraIOLock:
            try:
                subprocess32.call("gphoto2 --wait-event-and-download=100s",
                            cwd=self.imageDir + "/",
                            shell=True)
            except Exception as ex:
                template = "An exception of type {0} occurred. Arguments:\n{1!r}"
                message = template.format(type(ex).__name__, ex.args)
                print message
                print "Error copying files. Trying again"

    def loop(self):
        while not self.stopped:
            self.watch_files()

    def start_watching_files(self):
        self.stopped = False
        self.t = threading.Thread(target=self.loop)
        self.t.start()
```

J0hannB, 2017-07-12:
A new update: 

I left the camera on all night last night and ran `--wait-event-and-download` from the terminal. I forgot to adjust the camera's exposure for when the lights turned off, and, when they did, the size of the images being transferred dropped to about 2.5 MB. In this condition, the software worked flawlessly all night until I shut it off this morning after 15 hrs and more than 40,000 captures (meaning I probably cut the life of my shutter in half). I'm wondering if the issues I saw earlier had to do with USB transfer speed limitations. 

And now that I think of it, I do remember turning the camera off once after an error (and before I disconnected USB) and seeing a graphic indicating that the camera still had 20 or so images to transfer. 

Theoretically, USB 2.0 should allow 60MBps, but when I have timed copying large amounts of data from the camera over MTP in the past, I usually see something closer 15MBps. My current setup takes  .74 pictures per second, which means for USB to be the bottleneck, my pictures would have to be close to 20MB, right? That is definitely not the case as they usually average 7MB with the lights on.

msmeissn, 2017-07-12:
This might mean we could have missed some events and so left some images hanging.
What is the "capturetarget" configuration ? SDRAM or CARD?

In SDRAM mode it might leave images on the camera ram if events are missed, which could lead to blocking after a while. In CARD mode it would save all to SD Card, and that will be slower/.

For the speed I did not do measurements. There is the megabyte vs megabits thing, 15 megabits/s would be around 2 MegaByte/s? 

So the rate of 0.74 second/image would not be sustainable unless you really reduce the size.

J0hannB, 2017-07-27:
Oops, that was I typo. I meant 15 MBps (as in MegaBytes per second).

I can play around with setting capturetarget to CARD, but in case that is too slow, do you know of any way to prevent the camera from completely locking up when target is SDRAM and the buffer to tx to the PC is full? My experience has been that as soon as this happens, the camera will report 'busy' indefinitely and the only solution is to remove and reinsert the battery.

Thanks by the way for all of the help so far. Your info has been invaluable

J0hannB, 2017-07-28:
So, I played around with setting capturetarget to CARD, and I found that without fail, after 50 - 150 captures, gphoto2 just stops copying new files over to the PC. This happens regardless of the size of images or rate of capture. The solution is simply to restart the --wait-event-and-download process. The only problem is that I miss any photos that were taken while the command was not running. 

Is this some sort of an intentional timeout? If so, why don't I see it when capturetarget is set to SDRAM? Are there any workarounds?

Thanks again for the help!

msmeissn, 2017-07-30:
For locking up from SDRAM capture, you need to download the images. when it gets full it will be busy. I dont know any way around it. (You might be able to remove the images in SDRAM without downloading them, but this is not implemented and not that useful here.)

For the second comment ... I am not aware of any intentional timeout here. Does it report io errors or does it just hang and not report anything anymore? If the camera disconnects it should get IO Errors. 

Does the camera create new folders every 100 images? This might confuse some logic.

J0hannB, 2017-07-31:
But when the SDRAM buffer does fill up, how can I download the images? Once it happens, I can't do anything to the camera until I reboot it. 

When the gphoto2 stops copying images when the camera is in CARD mode, the failure is silent (i.e. I get no error message). It simply behaves as if no more pictures were being taken, even though the camera continues to capture new images. As I mentioned above, though, I don't need to reboot the camera to fix this. Running the command again is enough.

I believe the camera creates a new folder every 10000 pics, so I don't think that is causing the issue.

J0hannB, 2017-08-02:
It turns out the issue in CARD mode was my fault. For various reasons, I had switched to using the `subprocess.Popen()` method to run the command with `stdout` set to `PIPE` and then calling `wait()` to block until the process is finished. Since the `--wait-event-and-download` command generates a lot of output,  the stdout buffer was filling up after so many images. 

My solution is to constantly poll the Popen object and read the output each time instead of calling `wait()`

My question still remains about recovering from a full buffer in SDRAM mode, but CARD seems to be working well for me now

J0hannB, 2017-08-10:
My project requirements have changed, and I ran into the original err 70 issue again with my new code. Since I've caused this thread to go a little off-topic, I made a new issue [here](https://github.com/gphoto/libgphoto2/issues/195)

msmeissn, 2018-04-17:
can you retrest with libgphoto2 2.5.17 and open a new issue if it persists.

closed 2018-04-17 by msmeissn

eladberger, 2019-03-31:
J0hannB, are you still around? I'm super new to this and I'm trying to do the same thing

msmeissn, 2019-03-31:
This bug was fixed in recent releases, 2.5.22 should not have it.

(We nowdownload image in small pieces, as the full download occasionaly crashed the camera with error 70)

Dlahacz, 2019-08-14:
hi, exactly same issue here (Canon M50), After few hours of usage, it starts crashing (err 70 or PTP I/O error). I'm using latest version 2.5.23.

msmeissn, 2019-08-28:
@Dlahacz can you try getting a debug trace? what are you doing, downloading images or retrieving file lists?

Dlahacz, 2019-08-28:
Hi, I will try to, but when i debug level logging, fps of livepreview goes down and that is not what i want in pruduction for couple of hours. But I'll manage to somehow get it. It's happening while downloading images/preview.

Dlahacz, 2019-08-29:
I've created separated issue with debug file because it's seems to be little bit different. https://github.com/gphoto/libgphoto2/issues/440