gphoto/libgphoto2 #182: GoPro Hero 4 does not list files and folders correctly

mjd3, 2017-07-13:
Background:
I am trying to capture video on the GoPro (not using gphoto2), then transfer it to my machine using gphoto2's command line interface after I plug in a USB cable. I want to avoid having to take out the SD cards since they are often unreachable in the current setup that I have. Additionally, I am using another module that can remotely begin capturing for the GoPro; this module causes the camera to save the video files in a different folder than the usual GoPro folder. 

Problem:
If I do not use the other module to remotely capture, and instead just press the capture button with my finger, then after plugging in the camera I can see the correct output from gphoto2, e.g.
There is no file in folder '/'.                                                
There is 1 file in folder '/store_00010001'.
#1     Get_started_with_GoPro.url r-     1 KB text/html
There is no file in folder '/store_00010001/DCIM'.
There are 3 files in folder '/store_00010001/DCIM/100GOPRO'.
#2     GOPR3142.MP4               rd 18164 KB video/quicktime
#3     GOPR3143.MP4               rd 42664 KB video/quicktime
#4     GOPR3145.MP4               rd 16527 KB video/quicktime

However, with the module, files are saved in a folder with a different name at the same level as the 100GOPRO folder. If I run the same command as above, the output looks like this:
There is no file in folder '/'.                                                
There is 1 file in folder '/store_00010001'.
#1     Get_started_with_GoPro.url r-     1 KB text/html
There is no file in folder '/store_00010001/DCIM'.
There is no file in folder '/store_00010001/DCIM/100GOPRO'.

If there are files in both folders, the output is the same as the second case; neither folder is read correctly. If I then take the SD card out and read it with my computer, I can verify that the files are indeed there despite gphoto2 claiming they are not. Moreover, if I move the files from the strangely-named folder to the 100GOPRO folder and rename the files so that they match the normal GoPro filenames, then gphoto2 is able to list all of the files correctly again. Obviously doing that sort of file manipulation would require me to remove the SD card, which as I said above, I'd strongly prefer not to do.

Thanks for checking this out - it seems to be a very strange occurrence!

I have attached a log with the debug option as requested. Here is the output from running the summary (with the serial number removed):
$ gphoto2 --summary
Camera summary:                                                                
Manufacturer: GoPro
Model: HERO4 Black
  Version: HD4.02.03.00.00
  Serial Number: XXXXXXXXX
Vendor Extension ID: 0x6 (1.0)
Vendor Extension Description: microsoft.com: 1.0;  microsoft.com/WMPPD: 11.0; microsoft.com/WMPPD: 10.0; microsoft.com/WMDRMPD: 10.1;

Capture Formats: 
Display Formats: Apple Quicktime, Association/Directory, JPEG, HTML, Text, Undefined Type, Executable
Supported MTP Object Properties:
	Apple Quicktime/300d: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dc89/Duration dc08/DateCreated
	Association/Directory/3001: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name
	JPEG/3801: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dc08/DateCreated
	HTML/3005: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name
	Text/3004: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name
	Undefined Type/3000: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name
	Executable/3003: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_00010001:
	StorageDescription: GoPro MTP Client Disk Volume
	VolumeLabel: GoPro MTP Client SD slot
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Generic Hierarchical
	Access Capability: Read Only with Object deletion
	Maximum Capability: 31902400512 (30424 MB)
	Free Space (Bytes): 31820906496 (30346 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 100, step 1] value: 100% (100)
Date & Time(0x5011):(readwrite) (type=0xffff) '20200713T193944'
Property 0xd303:(read only) (type=0x2) 1
Synchronization Partner(0xd401):(readwrite) (type=0xffff) GoPro ('GoPro')
Friendly Device Name(0xd402):(readwrite) (type=0xffff) HERO4 Black ('HERO4 Black')
Session Initiator Info(0xd406):(readwrite) (type=0xffff) 'Unknown'
Perceived Device Type(0xd407):(read only) (type=0x6) 1

[gphoto_log.txt](https://github.com/gphoto/libgphoto2/files/1146715/gphoto_log.txt)

msmeissn, 2017-07-16:
the ptp filesystem retrieval is confused somehow. :/

the gopro reports object id 0x3 being present, but on querying it reports an error. 

There is no second folder being retrieved as far as I see, DCIM and the .lnk file in the root of store_00010001, the 100GOPRO in the DCIM directory. 

There error handling might be better here, but it would not show up. 

are the folders named similar on both storage things? this will probably overlap and conflict

mjd3, 2017-07-17:
The names of the two folders are 100GOPRO and 100G3D_L, both under the DCIM directory. Within the two folders, files are named according to the folder names. Thus, all of the files under the 100GOPRO directory are named GOPRXXXX.mp4 and all of the files under the 100G3D_L directory are named 3D_LXXXX.mp4, with XXXX being whatever number the video is. Hopefully that answers your question; let me know if you have any more!

mjd3, 2017-07-24:
I was able to fix this problem by using a patched version of the GoPro firmware which allows for mounting as USB mass storage.

closed 2017-07-24 by mjd3

msmeissn, 2017-07-25:
sadly it might still be a firmware issue somewhere. sorry, the 100G3D_L was not at all visible in your debug trace, so I suspect there is something either in the firmware or our understanding here amiss.
good to hear that you found a solution!

sops21, 2018-07-09:
@mjd3 What module are you using for capture?