gphoto/libgphoto2 #183: Canon M3 "Does not support generic capture"

samedmonds, 2017-07-15:
Hi, 
Despite updating to the latest release, it would appear that my Canon M3 is still unsupported. Any ideas as to what the problem might be? 

I'm connecting my M3 to a RaspberryPi and executing gphoto commands via ssh. 

Please see below for a screenshot of the error message when attempting to auto detect. 

[image]
[image]

msmeissn, 2017-07-15:
duplicat of https://github.com/gphoto/gphoto2/issues/84

closed 2017-07-15 by msmeissn

Sija, 2017-07-15:
@msmeissn typo: duplicat -> duplicate