gphoto/libgphoto2 #184: Make File Issue

samedmonds, 2017-07-16:
Hi, 

Am trying to update to 2.5.14 but running into an error when attempting to execute the Makefile. Please see below. 

[image]

msmeissn, 2017-07-16:
you need to run: 
./configure --prefix=/usr

to generate the Makefile out of Makefile.in 

after that suceeded run make

samedmonds, 2017-07-16:
Thanks for your help, Marcus. That worked a treat. 

I've since gone onto run `make` and then `sudo make install` but when attempting to --auto-detect my Canon M3, it still does not show up (please see below for output). Any suggestions? 

Sam. 

[image]

msmeissn, 2017-07-16:
It probably did not overwrite the system libgphoto2 library, you can see with above call that the new binary was found for the gphoto2 tool, but not for the library.

can you see if it installed libraries in /usr/local/lib instead ?

btw, there is an easy script for updating gphoto2, https://github.com/gonzalo/gphoto2-updater  , made by a contributor especiall for rasperry pi

samedmonds, 2017-07-16:
Thanks again, Marcus! 

I just tried the script you mentioned and it appears to be working!

msmeissn, 2017-07-25:
thx for reporting back!

closed 2017-07-25 by msmeissn