gphoto/libgphoto2 #185: Weird gp_camera_capture_preview behaviour with pipes

nmrmak, 2017-07-16:
Hi,

For some reason, gp_camera_capture_preview() will not work if the CameraFile passed to it is created from a pipe when using Canon EOS650D (but oddly the same code works with Nikon D7000).

Here is code that works on both cameras:
```
int ret;
CameraFile *pFile = NULL;
ret = gp_file_new(&pFile);
ret = gp_camera_capture_preview(pCamera, pFile, pContext);

if(GP_OK != ret)
{
    LOG("Unable to capture viewfinder frame in getViewFinderFrame()");
    gp_file_unref(pFile);
    return GP_ERROR;
}

ret = gp_file_get_data_and_size(pFile, (const char**)dataReturnValue, sizeReturnValue);
if(GP_OK != ret)
{
    LOG("Unable to get viewfinder frame data and size in getViewFinderFrame()");
    gp_file_unref(pFile);
    return GP_ERROR;
}
gp_file_unref(pFile);
return GP_OK;
```

And here is the entire function that works with Nikon, but blocks on gp_camera_capture_preview() with Canon. The pipe's file descriptor is passed as a parameter to this function, and is const int fd. The function is identical for Nikon and Canon, yet blocks for Canon.

```
CameraFile *pFile = NULL;

if(0 == fd)
{
    LOG("Cannot write to file descriptor because it is NULL");
    return GP_ERROR;
}

if(GP_OK != gp_file_new_from_fd(&pFile, fd))
{
    LOG("Could not create CameraFile from file descriptor");
    return GP_ERROR;
}

int ret = 0;
ret = gp_camera_capture_preview(pCamera, pFile, pContext);

if(GP_OK != ret)
{
    LOG("Unable to capture viewfinder frame in getViewFinderFrame()");
    gp_file_unref(pFile);
    return GP_ERROR;
}

return GP_OK;
```

In this case everything works fine until we hit gp_camera_capture_preview, at which point it hangs since it doesn't return from the function.

It is weird because I can get viewfinder frames from both camera types when not using pipes as you can see in the first example, but when I use pipes it won't work for Canon. I can create a workaround where I'll feed the pipe manually after getting a frame like I do in the first code snippet, but surely it should work this way too?

Thanks,
Nebojsa

nmrmak, 2017-07-16:
Hi again,

It's worse than I thought.

In the first snippet, I get frame size and frame data using gp_file_get_data_and_size, however it seems that this function returns an invalid pointer (not NULL though).

I've just tested and reading frame data works on Nikon but segfaults on Canon. This looks like a serious issue that I can't seem to be able to work around.

If you need any more data from me, please let me know.

Nebojsa

nmrmak, 2017-07-17:
The difference between Nikon and Canon is the frame size. Nikon frames are ~15 kB on my setup, and Canon frames are much larger when the frame is dark, but more importantly they are larger than 64 kB. Pipe writes block for writes of more than 65535 bytes on my machine (arm arch).

If I increase the pipe size using:
int pipe_sz = fcntl(viewfinderPipeDescriptor[1], F_SETPIPE_SZ, 524288);
then everything works (including gp_camera_capture_preview writing to a pipe directly, but this seems like an unrealistically large pipe size to me.

Maybe we could implement writing to pipes in chunks?

msmeissn, 2017-07-17:
gp_file_get_data_and_size does only work in FD mode if you use seekable files (files on disk). do you check the error return?

The preview function is just doing sequential writes, so the pipe size should not matter really.

This bug its not related to the livestream slow issue.

(I will be checking this after my vacation.)

nmrmak, 2017-07-18:
Hi,

I am checking the return value of gp_file_get_data_and_size, but it returns GP_OK. Does this mean that using this function with pipes is undefined behaviour, since pipes are not seekable? Maybe the documentation could be updated to reflect this if it is correct.

As for the preview function, (I haven't checked this) maybe it uses a blocking write which never completes since there is not enough room in the pipe. There might be some more logic needed to use non-blocking writes if this is the case (and if you decide that it's worth the effort)

Thank you for taking the time to reply from your vacation. Have a great time!

After you're back, I could try to help out with this some more if you'd like, just let me know.

Nebojsa

msmeissn, 2020-05-01:
I think the receiving end of the pipe needs to be fixed ... I mean if we write in stuff, it needs to be read out?

Not sure what is / was on your receiving end.

As of now, a number of people have this working, so  I hope its resolved :)

closed 2020-05-01 by msmeissn