gphoto/libgphoto2 #186: Using --set-config with EOS 750D always fails after first use

dcf21, 2017-07-17:
gphoto2 seems to have some issues talking to my Canon EOS 750D. If I use --set-config to set an exposure length, it works the first time, but always fails on all subsequent calls:

dcf21@thebe:~$ gphoto2 --set-config /main/capturesettings/shutterspeed=10
dcf21@thebe:~$ gphoto2 --set-config /main/capturesettings/shutterspeed=10      
                                                                               
*** Error ***              
/main not found in configuration tree.
*** Error (-1: 'Unspecified error') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config /main/capturesettings/shutterspeed=10

Please make sure there is sufficient quoting around the arguments.

The log file is attached...
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/1153739/my-logfile.txt)

When it comes to capturing images, it seems to work about 80% of the time, but sporadically returns a "could not capture" error.

My version numbers are as follows:

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.14         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.14         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking

Any thoughts?

arismelachroinos, 2017-12-09:
I also get 
*** Error ***
/main not found in configuration tree.
*** Error (-1: 'Unspecified error') ***

when i try to get the shuttercount from my 80D

fyrkrans, 2018-01-23:
I get that error message too when trying to get the shuttercount from a Canon 200D.  At first I thought it's simply not supported yet, but the model is mentioned in the list of supported cameras.

Can I help somehow?  Should I ask on the mailing list instead?

EliaCereda, 2018-01-24:
I get the same error from my Canon EOS 760D when changing various properties.

I found that the problem does not happen in `gphoto2 --config` or `gphoto2 --shell` nor if using `gphoto2 --set-config capture=1 --set-config <# actual config #> --set-config capture=0`

uladkasach, 2020-07-05:
Had the same issue. It looks like by default gphoto is in `capture=0` mode (which can be read as "capture_mode = read only")

By running
```
gphoto2 --set-config capture=1
```
It seems like we're setting the camera into not-read-only, which exposes the other settings.

msmeissn, 2020-07-06:
--set-config capture=1 would only be needed several years ago for Canon Powershot.

it is not needed in any new cameras currently. 

gphoto will get the camera into capture mode automatically if you do capture related activities.