gphoto/libgphoto2 #187: Best way to wait until PTP is not busy anymore?

jfgout, 2017-07-18:
First of all, let me say thank you to the people developing/maintaining libgphoto2! This is an amazing tool.

I'm using libgphoto2 to change settings on a camera but I trigger the capture with a serial cable. This means that sometimes I need to wait until the camera is done writing the last image before I can change a parameter. For example, if I'm shooting in burst mode and trigger for 2 seconds with the serial cable, it will take a second or two for the camera to finish writing the images after the 2 seconds have elapsed (the burst of 2 seconds takes a bunch of images, depending on shutter speed and other parameters).

Unfortunately, I can't use the gp_camera_wait_for_event function because there is an undetermined number of files to be written (so, just waiting for GP_EVENT_FILE_ADDED is not an option). My understanding is that GP_EVENT_CAPTURE_COMPLETE happens only when capture was triggered via usb.
One very dirty solution would be to wait for GP_EVENT_FILE_ADDED and make sure no such event happened in the last 1 second (assuming it takes no more than 1 second to write a file on the card, this would tell me that the camera is done writing the files from the burst). But I'd like to do something better and that would tell me when the camera is ready to take a new command from usb.

I tried also to simply call the function that sets a property on the camera (I'm using the set_config_value_string function from the examples) until it returns a non-error code. Problem is that on the second call, it always returns 0, even if it failed again (I call the function a first time, it return -130 / PTP device busy ; I call it again after 10ms (when the device is still busy) and it returns 0 - but it does not succeed in changing the value).

So, what would be the proper way to wait until the device is not busy any more?

Thanks for your help!

jf

msmeissn, 2017-07-24:
can you check what other events the camera returns? there might be some in the UNKNOWN events that indicate finishedness?  is this canon eos or nikon?

jfgout, 2017-07-25:
Thanks for your help. Unfortunately, I don know what to do with the 'UNKNOWN' events (there are plenty that happen during the time the camera is writing from the buffer to the memory card.
However, I did find a (dirty) fix to the problem.
The real problem comes from the fact that it looks like gphoto2 updates the setting to it's new value, even if the set function failed. For example, I try to set the exposure time to 1/1000 and it fails because the camera is busy, next time I try to read the value of the exposure time it will read 1/1000 (even though the camera is not set at 1/1000). Not sure if that's a bog or if that's how libgphoto2 is supposed to handle these things.
So, after I realized that, I implemented a very dirty fix to my problem. What I do is that I try to change the setting (say for example, set exposure speed to 1/250) and if it fails, I immediately try to set it to something else (for example, 1/1000). Like that, I can try again to change to the desired setting (1/250) and see if it fails or not (and just keep doing that until it succeeds).
Strangely enough, I found that typically it works after a few tries (I pause 100ms between each try) before the camera is done writing all the images from the burst on the memory card.
That's a dirty fix, but it seems to be doing the job for what I need :)

Oh, sorry I forgot to mention the cemra brand/model. It's a Canon 5dM2.

jf