gphoto/libgphoto2 #193: Bulb exposure on EOS cameras that have B on mode dial?

photog0411, 2017-08-08:
Hi, I got it working to do Bulb exposure on cameras that have "BULB" as a shutter speed in "M" mode, but is there a way to remotely do bulb exposure on higher-end EOS cameras that have a dedicated bulb mode on the mode dial, and no BULB exposure setting?

msmeissn, 2017-08-09:
see http://gphoto.org/doc/remote/

Newer Canon EOS DSLR:

The camera usually needs to be in 'M' mode.

gphoto2 --set-config shutterspeed=bulb
gphoto2 --set-config eosremoterelease=Immediate --wait-event=30s --set-config eosremoterelease="Release Full" --wait-event-and-download=2s

The eosremoterelease refers to the direct shutter button manipulation, so it can be used for different scenarios outside of immediate shutter releases.

nathanbird, 2017-08-09:
@msmeissn What about in my case where I'm attempting to accomplish this on a different platform? Do you know what hex codes associated with the "Immediate" and "Release Full" commands are? Thanks!

msmeissn, 2017-08-09:
did we not email today?

msmeissn, 2017-08-09:
eosremote release on PTP opcode is 0x9128 with 2 arguments, eosremoterelease off is 0x9129 with 1 argument.

The code to handle the config is here: https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/config.c#L5519

msmeissn, 2017-08-22:
see the code link i pasted above. _put_Canon_EOS_RemoteRelease(CONFIG_PUT_ARGS)  translates the gphoto2 configuration values for this option to PTP arguments.

ptp_canon_eos_remotereleaseon is 0x9128   (2 arguments passed)
ptp_canon_eos_remotereleaseoff is 0x9129 (1 argument passed)

closed 2017-08-22 by msmeissn