gphoto/libgphoto2 #195: Manually Access SD Card During Image Capture Results In Canon Error 70

J0hannB, 2017-08-10:
I had a similar issue that I posted about [here](https://github.com/gphoto/libgphoto2/issues/181), but as I learned new info, I no longer needed to use the commands that caused that issue in the first place, and my comments there went a little off-topic. 

My project requirements have since changed, and I am running into err 70 again. This time I narrowed the problem down a little more, so I figured it would be best to start a new issue. 

**Background:** 

I automate triggering of a Canon EOS 6D by an external source. I need gphoto2 to allow me to copy thumbnails of the most recent images as they are being taken. I tried mimicking the behavior of `--wait-event-and-download` by calling `--wait-event FILEADDED` and then calling `get-thumbnail [name or number] --skip-existing --filename=%f`. 

The camera is in continuous mode and takes bursts of 4 images every few seconds, so the card is usually pretty busy (`capturetarget` is set to CARD). 

**The Problem:**

After about 400 - 600 calls to --gphoto2 (both `--wait-event` and `--get-thumbnail`), the camera locks up and reports "err 70" on the LCD. The only recovery is to re-insert the battery. 

I tried 3 different cards (2 Sandisks and a Kingston), and deep formatted them on the camera. I tried 2 different camera bodies, 2 different lenses, and even running with no lens. The problem persisted, so I am reasonably confident this has something to do with gphoto2 or libgphoto2. 

I am also pretty sure that it is caused by trying to access the SD card while the camera is writing to it because I can cause the same issue by replacing `--get-thumbnail` with `--list-files`, but if I replace `--get-thumbnail` with `--abilities` for instance, or `--get-config` I never see the error. 

What intrigues me most is that running `--wait-event-and-download` while triggering is active works great. What is my method doing differently than `--wait-event-and-download` besides only downloading thumbnails? Is there a better way to get real-time thumbnails?

Again, I am making all of these calls from a python script using the subprocess module.  

Thanks in advance for the help!

msmeissn, 2017-08-11:
This error seems to popup when the filesystem on camera gets confused. As you say , parallel access to the filesystem from camera and gphoto2 seems to be a precondition

J0hannB, 2017-08-11:
Any idea why running `--wait-event-and-download` while the camera is capturing images works fine?

msmeissn, 2017-08-16:
i have no idea so far :(

J0hannB, 2018-02-25:
I know this issue has gotten a little old, but I wanted to give an update on the solution I settled on. 

To continuously copy thumbnails, I modified gphoto2 to include a `--capture-tethered-prev` option. I basically just copied some of the code for `--capture-tethered` but set it to only download thumbnails. So far it has worked great!

msmeissn, 2018-04-17:
I think I found the reason and fix now ... please test libgphoto 2.5.17.