gphoto/libgphoto2 #196: Olympus E400 - failed: 'PTP Timeout'

Grauen, 2017-08-11:
I connected my Olympus E400 and set the mode to "Remote Control" - the camera is recognized by gphoto2 and I also can successfully select the camera using
`gphoto2 --auto-detect`

When I try to run the --summary command or --capture-image command there will be a timeout.

log running command
`sudo gphoto2 --debug --debug-logfile=gphoto2_olympus_e400_capture_image-sudo.log --capture-image
`

`0.000068 main                        (2): BITTE IMMER FOLGENDE ZEILE EINFÜGEN WENN DEBUG MESSAGES AN DIE MAILINGLISTE GESCHICKT WERDEN:
0.000116 main                        (2): gphoto2 2.5.9
0.000135 main                        (2): gphoto2 wurde mit den folgenden Optionen übersetzt:
0.000143 main                        (2):  + gcc (C compiler used)
0.000146 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000148 main                        (2):  + exif (for displaying EXIF information)
0.000151 main                        (2):  + no cdk (for accessing configuration options)
0.000153 main                        (2):  + aa (for displaying live previews)
0.000155 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000158 main                        (2):  + readline (for easy navigation in the shell)
0.000160 main                        (2): libgphoto2 2.5.9
0.000163 main                        (2): libgphoto2 wurde mit den folgenden Optionen übersetzt:
0.000166 main                        (2):  + all camlibs
0.000168 main                        (2):  + gcc (C compiler used)
0.000170 main                        (2):  + ltdl (for portable loading of camlibs)
0.000173 main                        (2):  + EXIF (for special handling of EXIF files)
0.000175 main                        (2): libgphoto2_port 0.12.0
0.000178 main                        (2): libgphoto2_port wurde mit den folgenden Optionen übersetzt:
0.000180 main                        (2):  + gcc (C compiler used)
0.000182 main                        (2):  + ltdl (for portable loading of camlibs)
0.000185 main                        (2):  + USB (libusb1, for USB cameras)
0.000187 main                        (2):  + serial (for serial cameras)
0.000189 main                        (2):  + no resmgr (serial port access and locking)
0.000191 main                        (2):  + no ttylock (serial port locking)
0.000200 main                        (2):  + no lockdev (serial port locking)
0.000203 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000205 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000208 main                        (2): invoked with following arguments:
0.000210 main                        (2):   --debug
0.000213 main                        (2):   --debug-logfile=gphoto2_olympus_e400_capture_image-sudo.log
0.000215 main                        (2):   --capture-image
0.000228 load_settings               (2): Creating gphoto config directory ('/home/wolf/.gphoto')
0.000250 load_settings               (2): Loading settings from file '/home/wolf/.gphoto/settings'.
0.000266 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000271 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.0'...
0.000313 foreach_func                (2): Called for filename '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.0/disk'.
0.000440 gp_port_library_list        (2): found fstab fsname UUID=fee70d70-1a6b-40be-9d89-2858b5739cef
0.000489 gp_port_library_list        (2): found mtab fsname sysfs
0.000494 gp_port_library_list        (2): found mtab fsname proc
0.000498 gp_port_library_list        (2): found mtab fsname udev
0.000505 gp_port_library_list        (2): found mtab fsname devpts
0.000508 gp_port_library_list        (2): found mtab fsname tmpfs
0.000515 gp_port_library_list        (2): found mtab fsname /dev/sdc2
0.000519 gp_port_library_list        (2): found mtab fsname securityfs
0.000523 gp_port_library_list        (2): found mtab fsname tmpfs
0.000529 gp_port_library_list        (2): found mtab fsname tmpfs
0.000535 gp_port_library_list        (2): found mtab fsname tmpfs
0.000543 gp_port_library_list        (2): found mtab fsname cgroup
0.000547 gp_port_library_list        (2): found mtab fsname pstore
0.000554 gp_port_library_list        (2): found mtab fsname cgroup
0.000557 gp_port_library_list        (2): found mtab fsname cgroup
0.000584 gp_port_library_list        (2): found mtab fsname cgroup
0.000588 gp_port_library_list        (2): found mtab fsname cgroup
0.000591 gp_port_library_list        (2): found mtab fsname cgroup
0.000594 gp_port_library_list        (2): found mtab fsname cgroup
0.000597 gp_port_library_list        (2): found mtab fsname cgroup
0.000601 gp_port_library_list        (2): found mtab fsname cgroup
0.000604 gp_port_library_list        (2): found mtab fsname cgroup
0.000607 gp_port_library_list        (2): found mtab fsname cgroup
0.000610 gp_port_library_list        (2): found mtab fsname systemd-1
0.000613 gp_port_library_list        (2): found mtab fsname debugfs
[... 1485 lines cut ...]
80.926078 _close_async_interrupts     (2): checking: transfer 6:0x562c9d242058 status 0
80.926084 _close_async_interrupts     (2): checking: transfer 7:0x562c9d254958 status 0
80.926090 _close_async_interrupts     (2): checking: transfer 8:0x562c9d254ba8 status 0
80.926095 _close_async_interrupts     (2): checking: transfer 9:0x562c9d249748 status 0
80.926113 _cb_irq                     (2): 0x562c9d231258 with status 3
80.926124 _cb_irq                     (2): 0x562c9d26d978 with status 3
80.926132 _cb_irq                     (2): 0x562c9d249528 with status 3
80.926139 _cb_irq                     (2): 0x562c9d21bf78 with status 3
80.926147 _cb_irq                     (2): 0x562c9d26e608 with status 3
80.926154 _cb_irq                     (2): 0x562c9d26e858 with status 3
80.926161 _cb_irq                     (2): 0x562c9d242058 with status 3
80.926167 _cb_irq                     (2): 0x562c9d254958 with status 3
80.926174 _cb_irq                     (2): 0x562c9d254ba8 with status 3
80.926181 _cb_irq                     (2): 0x562c9d249748 with status 3
80.926466 gp_context_error            (0): An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available
80.926950 gp_camera_free              (2): Freeing camera...
80.926960 gp_port_free                (2): Freeing port...
80.926966 gp_port_close               (2): Closing port...
80.927252 gp_filesystem_reset         (2): resetting filesystem
80.927272 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
80.927278 gp_filesystem_lru_clear     (2): fscache LRU list already empty
80.927284 delete_all_folders          (2): Internally deleting all folders from '/'...
80.927290 lookup_folder               (2): Lookup folder '/'...
80.927296 lookup_folder               (2): Found! / is 0x562c9d2123e0
80.927302 recurse_delete_folder       (2): Recurse delete folder 0x562c9d2123e0//`

msmeissn, 2017-08-11:
The olympus is using a backchannel  over  the mass storage USB/SCSI code.

But it needs access to it, so you need to have the camera un-mounted. If it is mounted, umount it.

Grauen, 2017-08-11:
I compared the output of df and mount and there was no difference when you put the camera to 'control' mode. When I set the camera to 'storage' mode it will be mounted but you cannot use gphoto2 when the camera is set to storage mode. If I unmount the camera in storage mode, gphoto2 will not detect the camera.

Grauen, 2017-08-11:
I just executed
`gphoto2 -a 

Abilities for camera             : Olympus E series (Control)                  
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Capture not supported by the driver
Configuration support            : no
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes
`

msmeissn, 2017-08-16:
-a does not query the camera ;)

could you attach the debuglog as file, it seems slightly corrupted by pasting (due to the XML tags)