gphoto/libgphoto2 #197: gphoto2 --capture-image doesn't record image to memory card in EOS 5D mk.III

vectorsigma, 2017-08-21:
Using `gphoto2 --auto-detect --debug --debug-logfile=eos5dmk3-gphoto2.5.14.log --capture-image` results in no image captured to my memory card(s).  Live view on or off, the result is the same.  Dual memory cards or single memory card, the result is the same.

I'm using 2.5.14, downloaded, compiled and installed today.  The bug existed in 2.5.11 as well, which I was using this morning..  during the [2017 eclipse](https://eclipse2017.nasa.gov/)..  So I now I have zero photos from 2 of 4 cameras. :(

The frame counter stays the same every time I run that command, with no variance.  Using `--capture-image-and-download` will download the image successfully.

Here's the output of two successive runs:
```
$ gphoto2 --auto-detect --debug --debug-logfile=eos5dmk3-gphoto2.5.14.log --capture-image
Model                          Port                                            
----------------------------------------------------------
Canon EOS 5D Mark III          usb:002,027     
New file is in location /capt0000.cr2 on the camera                            

$ gphoto2 --auto-detect --debug --debug-logfile=eos5dmk3-gphoto2.5.14.log --capture-image
Model                          Port                                            
----------------------------------------------------------
Canon EOS 5D Mark III          usb:002,027     
New file is in location /capt0000.cr2 on the camera
```
The camera is a 5D Mark 3, firmware rev 1.3.4.

The original failure case occurred with this invocation:
`gphoto2 --frames 360 --interval 20 --auto-detect --capture-image`

This also failed to correctly capture an image on the Rebel T5 as well as the 5D Mk. III.

However, the 20D and 300D work fine with that syntax.

[Debug log file](http://www.deepgroove.org/eos5dmk3-gphoto2.5.14.log.zip), too big to attach at ~50MB.

I'm very happy to help test and get this issue resolved, so no one else has to go through this.

vectorsigma, 2017-08-22:
I own the 5D, but not the T5, so I will only be able to help test the former.  Any pointers you can offer to help me test and narrow down the issue?

msmeissn, 2017-08-22:
the /capt... filenames are used in a mode where the image iis transfered directly from camera memory to the computer.

Change the "capturetarget" setting of gphoto2 to "card" to leave the images on the SD Card.

gphoto2 --set-config capturetarget=1

vectorsigma, 2017-08-23:
I will try that tonight and see if that makes things work as expected.  Thanks for the quick reply!

vectorsigma, 2017-08-24:
I tried that, as follows, and it worked:
```
$ gphoto2 --set-config capturetarget=1

$ cat .gphoto/settings
gphoto2=port=usb:001,007
gphoto2=model=Canon EOS 5D Mark III
libgphoto=cached-images=2
ptp2=capturetarget=card

$ gphoto2 --auto-detect --capture-image
Model                          Port
----------------------------------------------------------
Canon EOS 5D Mark III          usb:001,007
New file is in location /store_00010001/DCIM/101EOS5D/DG8C0337.CR2 on the camera
New file is in location /store_00020001/DCIM/101EOS5D/DG8C0337.CR2 on the camera
```

Can I assume that having to apply the extra setting is a bug?  A naive reading of the section for `--capture-image` would lead most people to believe (as I did) that using the `--capture-image` flag explicitly states that the image is stored on the card, without additional settings requirements.  (This works as expected for the Canon 300D and 20D, but I have no other models to test.)

If so, what else can I do to help?

msmeissn, 2017-08-28:
well, it is more a default setting or documentation issue ... so far the default is SDRAM, but it probably would be more intuitive to have the default being CARD.

ayk-tonesoft, 2018-04-09:
OMG! I spent two days trying to fix this problem in my setup... I was barking on the wrong tree because the following log message made me thing the problem was in USB settings:

```
0.490623 ptp_usb_getresp [usb.c:466] (0): PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
0.490695 camera_unprepare_canon_eos_capture [config.c:557](0): 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)
```

Does this setting `--set-config capturetarget=1` persist or I need to reapply it after each reboot?

Thank you!

msmeissn, 2018-04-15:
it persists (it is stored in ~/.gphoto/settings )

closed 2018-04-15 by msmeissn

vectorsigma, 2019-11-17:
so did the default change to CARD now?