gphoto/libgphoto2 #198: Canon FlexiZone AF support

evertheylen, 2017-08-23:
I'm using a Canon 1300D, which has a feature called FlexiZone Autofocus (like many Canons I suspect). Basically, it's a rectangle on your screen that you can move around with the arrows (comparable to touch to focus on touch screen devices). Here's an example:

[image]

I'm wondering whether I can control this through libgphoto. I know how to trigger the autofocus, but not how to move the rectangle. I've tried `gphoto --list-all-config` before and after moving the rectangle and diff'ed the output, without success.

At this point, I'm thinking it's probably not exposed through USB. However, could it be that there is some vendor-specific extension to PTP that would allow this? With some guidance, I'd be very happy to implement this myself.

For reference, I added the output of `--list-all-config`: [canon1300D_config.txt](https://github.com/gphoto/libgphoto2/files/1246034/canon1300D_config.txt)

Great library btw ;)

msmeissn, 2017-08-24:
can you capture debuglog before and after.

we might just not be decoding the values yet.

evertheylen, 2017-08-24:
Thanks for your response! Here are the logs, captured with `gphoto2 --debug --debug-logfile=square-location-debug.txt --debug-loglevel=all --list-all-config`:

  1. [square-topleft-debug.txt](https://github.com/gphoto/libgphoto2/files/1248097/square-topleft-debug.txt)
  2. [square-mid-debug.txt](https://github.com/gphoto/libgphoto2/files/1248098/square-mid-debug.txt)
  3. [square-bottomright-debug.txt](https://github.com/gphoto/libgphoto2/files/1248101/square-bottomright-debug.txt)

evertheylen, 2017-08-28:
I tried to do the same diff trick with the debug logs (after stripping away the timestamps), but I didn't find anything that could resemble the flexizone rectangle location. Is there another layer of code where the values/settings might be discarded, or do the debug logs already show every possible setting available to us?

msmeissn, 2017-08-30:
i also diffed it, no diff excepting the d17c aka EOS_UTCTime.

either we need to request it seperately, or there might be commands handling it.

can you try to gfet a trace of this on windows?

evertheylen, 2017-08-30:
I do have a Windows computer available, but I'm unsure of what you want. Is it related to the windows SDK Canon provides, or do I just run the same command as before but on windows?

msmeissn, 2017-08-30:
The idea would be to install USB snooper (SnoopyPro or similar) and run the windows software to do this and capture its USB trace. Not sure if modern Windows still allow this kind of snooping, have not done this for any version after XP.

Atterratio, 2017-09-09:
Hmmm open source "digiCamControl" can do this with my 1200D, tomorrow I try catch necessary USB log.

Atterratio, 2017-09-10:
I confirm there is no change in the output of the `gphoto2 --auto-detect --list-all-config` command before and after the Canon FlexiZone AF changes of the standard camera tools.
I add USB dumps of setup FlexiZone AF on three Canon cameras: EOS 1200D, EOS 650D and EOS 7D, by two camera control programs: EOS Utility 2(offisial Canon software) and digiCamControll(windows open source camera control software). All dumps get on live wiev mode, becouse its nesesary to change FlexiZone AF and contain many unnesary data. Dumps get by USBPcap utility and compatable with Wireshark. Hopefully this will help add this feature to the library.
Dumps archive to big to directly paste, add link to google drive: https://drive.google.com/open?id=0B3iazkmHO2EQYWdZWVNKQmp4b1U

P.S.: maybe better way contact with digiCamControl developers and find out how they do this feature?

msmeissn, 2017-09-10:
I looked into the 1200D traces...

It basically does what we already also can do with the zoomposition variable:

--set-config zoomposition X,Y      will probably move the focus area. 

digicamControl also does cancel the liveview autofocus before
 --set-config cancelautofocus=1 

(digicamControl seems to link the EOS SDK)

Atterratio, 2017-09-11:
Hmmm It's look like, but:
```
Canon EOS 1200D                usb:001,006     
gphoto2: {/} /> get-config /main/actions/eoszoomposition
Label: Canon EOS Zoom Position
Type: TEXT
Current: 0,0
gphoto2: {/} /> set-config /main/actions/eoszoomposition=500,600
gphoto2: {/} /> get-config /main/actions/eoszoomposition
Label: Canon EOS Zoom Position
Type: TEXT
Current: 0,0
```
Look like it's right command because AF zone go to top left corner but not do nothing else.

msmeissn, 2017-09-11:
We are currently not getting the returned zoomposition as setting is an actual function call.

It is probably encoded in the various datablobs returned in the preview capture, although I do not see it on first glance.

msmeissn, 2017-09-11:
need to check the other traces to see if there is a AF rectangle size change in it, but if you did not , can you also attach one trace where you changed the size ofthis rectangle?

Atterratio, 2017-09-11:
It's look like not changing size only position, at least through camera tool, through the official software or through another application I can not change its size just the position.
If you want I can do video capture how it look like and get paralel USB dump this can help at least slightly correlate my actions through the software with the USB traffic.

Atterratio, 2017-09-13:
I'm not sure whether this will be useful, but here [video](https://drive.google.com/open?id=0B3iazkmHO2EQUmduZDNTWTM1TE0) with the change of the AF zone through the official program on the EOS 1200D c with an approximate timestamp and here [USBDump](https://drive.google.com/open?id=0B3iazkmHO2EQelpaU1NIWTJVT0U).

avisecag, 2019-03-20:
Just thought I'd leave this here, since I just had to implement this very feature in our gphoto2-based application.
You *can* set the FlexiZone with the eoszoomposition widget (though not retireve them, which was mentioned above). The condition is that you are in live view (viewfinder=1) while doing so. cancelautofocus=1, which was mentioned above, is not mandatory.

The biggest problem that confounded me was that the input is a bit finicky. The display is divided into abstract units, with the range going from 1,1 (upper left corner) to 3900x2500 (lower right corner).
These seem to be the maximum values that bring the flexizone rectangle to the lower left screen.
The confounding thing is that invalid values don't do anything without producing an error. Invalid values are any negative values, zero, and depending on x or y they arbitrarily start somewhere above 4600/3000. 
In other words, there are some values that are still valid and will set the zone to the edges of the display when they were somewhere else before, but above some threshold they stop to do anything and the rectangle stays where it is.
THis little detail confused the hell out of me while testing, no need for anybody else to get confused by it.

avisecag, 2019-03-26:
Ok, the above only works on the 1200D. On the 2000D I can't get the flexizone to move.