gphoto/libgphoto2 #201: gphoto2 caused error 70 on 5d mk2

headlights, 2017-09-23:
I did run config codes i did write for a 40D to the 5d mk2

now i have error 70 .  Did  remove both batteries and did format the card. 

I am a newb but does anybody can help me to fix the error 70 

atm i cannot make contact with the camera using the usb port as it seems error 70 does block all access

any tips / help

headlights, 2017-09-24:
update  have now this error *** Error ***              
Canon EOS Half-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

headlights, 2017-09-24:
update 2:

Camera is acessable in C1, C2,C3,but gives in all other functions / coises error 70. 
in C1 etc the camera is also acessable by Entangle , so i do assume also with gphoto , but do not dare to scheck that. 

anybody any clue?

headlights, 2017-09-24:
found the solution aka  for me, and i  do think for many with error 70 due to writing invalid data to the camera.

the c1, c2 c3 options are NOT affected. and there is a way to copy that data back to the main mem. So there is real way to reset the camera. 

see: http://www.reikanphotography.co.uk/blog/2011/11/Canon-EOS-Cameras-Recovering-from-Err-70

msmeissn, 2017-09-24:
question is really which config variable triggered this when setting.

i have some repors of error 70 , but so far no clear indication what actually causes it :(

headlights, 2017-09-24:
nor do i. from memory.  i did run valid 40D settings to the 5 D mk II , i did go to console and did run take image with download and then take image without download. Then the error 70 did occur. I know this info is very vaque but all i have for you   I am sure gphoto dd cause it in comby with the 5d mk II softwre with is much more strict then teh 40 D  as it seems to me.  

sry all i can give to you for now. 

for me i dare not to touch ghoto2 with my 5d mkii anymore. which is a shame as gphoto2 is such an awesome project imo.

ArXen42, 2019-08-28:
Found this issue googling for "err 70" problem solution. I just witnessed two 5D MkII being bricked by setting incorrect image format from Canon EDSDK.

Maybe this description will help somebody:

1) My code read ImageQuality property (0x00000100) from camera and save it to some variable
2) Then performs "probing" all possible formats by trying to write them into camera. If camera returns error - the format is considered to be unsupported.
3) After probing, previously saved value is restored.

On step three the camera throws "DEVICE_BUSY" error and becomes Err70 brick. Right now I have no idea what exact value causes that problem and how to fix it. Will comment if there will be new details.

msmeissn, 2019-10-20:
What I found is that downloading large images caused it, especially RAW images.

In current libgphoto2 versions I split the download using GetPartialObject to avoid this.

Please try libgphoto2 2.5.23