gphoto/libgphoto2 #202: gphoto2 16 camera limit

daboi, 2017-09-25:
Hi,
I am trying to control 96 Canon cameras with gphoto2 but I cannot get more than 16 cameras displayed on auto-detect. I tried to use different USB ports of my mac but it did not recognize more than 16 cameras. Any suggestions for listing more than 16 cameras on one mac ?

msmeissn, 2017-09-26:
is "lsusb" showing more than 16 cameras?

This is kind of weird, libgphoto2 has no limitation there as far as I see.

daboi, 2017-09-26:
When I connect 12 cameras to one port. lsusb shows all 12 cameras. When I connect other 14 cameras alone, lsusb also shows 14 cameras. But when I connect both hubs it shows no camera at all! I use Canon 1200D cameras. What can be the reason for this ? I tried raspbian and macosx sierra, both act same.

msmeissn, 2017-09-27:
This might be a power related issue, the hubs are probably powered? or some root hub issue.

is there anything error / weird usb related in "dmesg" when plugging in both?

daboi, 2017-09-28:
Yes it says USB disconnected. and lists several error messages. It might be my usb hubs but there are a lot to change. So currently I will stick with 16 camera per controller.

msmeissn, 2017-10-14:
not much libgphoto2 can do here, sorry

closed 2017-10-14 by msmeissn