gphoto/libgphoto2 #204: No confing named "capturetarget" in gphoto2 --list-config

vacky11, 2017-10-02:
Hello,

I am using Sony alpha7r camera. My main problem is that I am not able to capture and save image on sd-card of the camera using command gphoto2 --capture-image. 

```
nvidia@ubuntu-2:~$ gphoto2 --capture-image
New file is in location /capt0000.jpg on the camera                            
New file is in location /capt0000.arw on the camera 
```

Then I found out on some gphoto threads that this can be fixed by changing the config of "capturetarget" to SDCARD. But I dont see such option in gphoto2 --list-config

```
nvidia@ubuntu-2:~$ gphoto2 --list-config 
/main/actions/autofocus                                                        
/main/actions/capture
/main/actions/bulb
/main/actions/movie
/main/actions/opcode
/main/status/serialnumber
/main/status/manufacturer
/main/status/cameramodel
/main/status/deviceversion
/main/status/vendorextension
/main/status/batterylevel
/main/imgsettings/imagesize
/main/imgsettings/iso
/main/imgsettings/colortemperature
/main/imgsettings/whitebalance
/main/capturesettings/exposurecompensation
/main/capturesettings/flashmode
/main/capturesettings/f-number
/main/capturesettings/imagequality
/main/capturesettings/focusmode
/main/capturesettings/expprogram
/main/capturesettings/aspectratio
/main/capturesettings/capturemode
/main/capturesettings/exposuremetermode
/main/capturesettings/shutterspeed
/main/other/5004
/main/other/5005
/main/other/5007
/main/other/500a
/main/other/500b
/main/other/500c
/main/other/500e
/main/other/5010
/main/other/5013
/main/other/d200
/main/other/d201
/main/other/d203
/main/other/d20d
/main/other/d20e
/main/other/d20f
/main/other/d210
/main/other/d21c
/main/other/d211
/main/other/d213
/main/other/d21e
/main/other/d21b
/main/other/d21d
/main/other/d217
/main/other/d218
/main/other/d219
/main/other/d212
/main/other/d214
/main/other/d215
/main/other/d2c1
/main/other/d2c2
/main/other/d2c3
/main/other/d2c8
/main/other/d2c5
/main/other/d2c7
```

Please help!

msmeissn, 2017-10-02:
Capturing to sd card is (to our knowledge) not possible on Sony Alpha series cameras. This is a firmware limitation.

vacky11, 2017-10-02:
Thanks for quick response

closed 2017-10-02 by vacky11