gphoto/libgphoto2 #205: Not working on MacOS

mccodesurfer, 2017-10-06:
Still cannot connect to my camera (Pentax) on a Mac. (It works with my camera (Pentax) on linux but not Mac)(killall PTPCamera doesn't help)

Is there any effort to get gphoto2 working on the Mac?

mccodesurfer, 2017-10-06:
There used to be this tool. https://github.com/mejedi/mac-gphoto-enabler
Don't know whats become of it

msmeissn, 2017-10-14:
The pentax uses SCSI pass through.  We currently implement this using Linux methods currently.

I can add something like in the sierra driver where I emulate this over libusb too.

But right now, it is not possible on Mac.

(Or is there a userspace interface to SCSI devices?)

mccodesurfer, 2017-10-18:
Do you mean the Mac uses SCSI pass through? This sounds like both the Mac and the Pentax have two strikes against them!

I don't really understand the issues. But I know the macOS is capturing the camera whether its in PTP or MSC mode. Kstars was using killall PTPCamera to free up the interface but it still did not have proper access to the camera. There was a problem with the Pentax on StellarMate (linux app) also, but those people found a solution. Thats all I know.

mccodesurfer, 2018-02-07:
Hi @msmeissn , I wrote a USB driver of sorts for the Mac. I was reviewing you comment above and realized that maybe with the driver I wrote you could have a channel to the control the camera. I stopped working on it because I had no information on the control API for the camera. but it seems you have that. Can I send you my Xcode project to inspect and see if its useful. (I've written an iOS app available on iTunes Store but this is my first attempt at Mac development.)

lwei2, 2020-10-14:
Hi,how do I send SCSI instructions to a plugged USB device on a MacOS?