gphoto/libgphoto2 #206: eosremoterelease is not working anymore for Canon DSLRs

knro, 2017-10-06:
Everything worked fine up to 2.5.9, when I updated to latest build from Github (2.5.15.1), I cannot control my Canon 600D camera. Dial is set to M and mode to Bulb. When I switch back to libgphoto2 v2.5.9, I can capture just fine without any issues. This is the command I used:

```
gphoto2 --debug --debug-logfile=gphoto.txt --set-config eosremoterelease=Immediate --wait-event=30s --set-config eosremoterelease="Release Full" --wait-event-and-download=2s
                                                                               
*** Error ***              
The eosremoterelease widget is not configurable.
Waiting for 30 seconds for events from camera. Press Ctrl-C to abort.

```
Attached complete log file:

[gphoto.txt](https://github.com/gphoto/libgphoto2/files/1364065/gphoto.txt)

msmeissn, 2017-10-07:
I might have made a check condition too strict. 

Can you fetch current git again and add debuginfo, I added some more debugging in the relevant function.

knro, 2017-10-07:
Sure, same error, see attached.

[gphoto.txt](https://github.com/gphoto/libgphoto2/files/1364547/gphoto.txt)

msmeissn, 2017-10-07:
This is weird. the same function returns error which was returning success now.

I did another fix that showed as error, can you pull again and try again?

knro, 2017-10-07:
Sure, here is the log:
[gphoto.txt](https://github.com/gphoto/libgphoto2/files/1364676/gphoto.txt)

msmeissn, 2017-10-07:
ok, there was a bad regression in string unpacking leading to this I think.

Can you pull git again and see if it is now fixed?

knro, 2017-10-07:
Yes it's fixed now!! Thanks a lot!!

closed 2017-10-07 by knro

Sija, 2017-10-07:
@msmeissn `2.5.15.1` hotfix release would be nice.

msmeissn, 2017-10-09:
i am looking for a free evening this week to do one.