gphoto/libgphoto2 #209: Nikon D750 -> Playback is not currently available

song314, 2017-10-19:
I have both Nikon D750 & Canon 80D, only 80D can us "playback" when I connect the USB

How to solve，cloud anybody help me. Pleeeeeeeeeeeeeease.

Thaaaaaaaaaaaaank a lot.

msmeissn, 2017-10-21:
you mean that the camera buttons work when USB is plugged in?

This is a bit of limitation of the Nikons, they lock the camera buttons/UI when plugged in.

song314, 2017-10-23:
Thanks, I found the doc about MTP for Nikon, and it works well, But where can I found the doc for Canon, I only found EDSDK, there is no docs for MPT

2.4	Application Mode
The camera is switched to the application mode by the ApplicationMode property.  At this time, the StoreRemoved event is passed.
In the application mode by the camera mode, the image playback and deletion can be performed on the camera.  However, the live view on the camera cannot be executed.

The host, which acquires the events by using the GetEvent command (subsection 6.2.2.6), can acquire all the events generated in the application mode.
The host, which acquires the events by using the Interrupt IN transfer, can acquire only the DevicePropChanged event of the standard property.

hyperparameters, 2024-09-12:
@song314 can you explain how did you solve this issue? what event/config do I need to set in order to access the playback on my DSLR