gphoto/libgphoto2 #210: gp_camera_get_single_config hangs randomly with Canon EOS 7D

miks, 2017-10-22:
Problem: gp_camera_get_single_config hangs indefinitely
How to repeat: I use libgphoto2 in my software, but it's easy to repeat this error with multiple `gphoto2 --get-config capturetarget` calls. First 2-3 will succeed, but after that gphoto2 will stay reading config forever. 

With `--debug` option it's seems that there is some kind of indefinitely reading going on. Also CPU usage is unusual high for gphoto2.
When canceling gphoto2 with ctrl+c, next config read will fail with "PTP Timeout" and only way to solve timeout problem is to reconnect camera usb cable.
Please, see attached log file.
[get_config.log](https://github.com/gphoto/libgphoto2/files/1404744/get_config.log)

Camera: Canon EOS 7D
Camera firmware: 2.0.3
libgphoto2 version: 2.5.16 
OS: High Sierra (but I can replicate error also on Ubuntu and ARM hardware)

miks, 2017-10-23:
Looks like EOS firmware update to 2.0.6 fixed my problem.

closed 2017-10-23 by miks

msmeissn, 2017-10-29:
good to hear the new firmware helps! :)