gphoto/libgphoto2 #212: Nikon D7000 photo resolution

Tyranus333, 2017-11-02:
Hi,
Python incompetant here (speak slowly with small words).
Using a Nikon D7000 as the camera in a photobooth on a raspberry pi which I leached thanks to /reutarbal I've thrown this question up there in case I'm missing something obvious. 
The D7000 is capapble of taking 300dpi images, which it does when triggered manually gphoto2 --capture etc. however when the camera is triggered by the photobooth with no mention or settings that I can see restricting the dpi it returns with a 19MB file and 96dpi image.
Is there a file somewhere that can be changed to fix this to be a standard 5MB JPEG and 300dpi image?

Thanks,

msmeissn, 2017-11-04:
Does this mean the imagesize parameter? the d7000 has 4928x3264 , 3696x2448 and 2464x1632

or the jpeg quality setting?

I do not remember a specific DPI setting and 300dpi would get a much higher size than 96DPI

Tyranus333, 2017-11-05:
I have specified the resolution of 4928x3264 but it made no difference on the reulting images.
I did not see anything about a JPEG quality setting in the script, but I guess that could be it. If I run from the bash command gphot2 --capture-image (or whatever the actual command is) it will capture it at 4928x3264 300dpi.