gphoto/libgphoto2 #218: Intermittent PTP I/O Error

kd7lxl, 2017-12-13:
I have a D3100 that's been working fine with Gphoto2 for a few months. All of a sudden it is giving intermittent PTP I/O Errors. Commands are failing about 90% of the time now. It seems to just give up in the middle of a session. libusb returns `LIBUSB_ERROR_OTHER`. I suspect there's nothing you can do about this, but just incase it is helpful for improving error recovery code or something, here is a log.

#### Interesting snippet:
```
2.417098 ptp_list_folder             (2): Listing ... 
2.418338 ptp_usb_sendreq             (2): Sending PTP_OC 0x1007 / Get object handles request...
2.419672 gp_port_write               (3): Writing 24 = 0x18 bytes to port...
2.421506 gp_port_write               (3): Wrote   24 = 0x18 bytes to port: (hexdump of 24 bytes)
0000  18 00 00 00 01 00 07 10-0f 00 00 00 01 00 01 00  ................
0010  00 00 00 00 00 00 19 11-                         ........        

2.422856 ptp_usb_getdata             (2): Reading PTP_OC 0x1007 / Get object handles data...
2.424884 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
2.464664 gp_libusb1_read [libusb1.c:559](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Other error (-99)
2.465080 gp_port_read [gphoto2-port.c:437](0): Reading 1024 = 0x400 bytes from port failed: Error reading from the port (-34)
```
[libusberror.log](https://github.com/gphoto/libgphoto2/files/1557121/libusberror.log)

under-score, 2019-01-06:
+1
Was working for a long time with identical hardware. 

```
0.000017 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000076 main                        (2): gphoto2 2.5.20
0.000085 main                        (2): gphoto2 has been compiled with the following options:
0.000092 main                        (2):  + clang (C compiler used)
0.000098 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000104 main                        (2):  + no exif (for displaying EXIF information)
0.000111 main                        (2):  + no cdk (for accessing configuration options)
0.000117 main                        (2):  + no aa (for displaying live previews)
0.000123 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000129 main                        (2):  + readline (for easy navigation in the shell)
0.000139 main                        (2): libgphoto2 2.5.20
0.000145 main                        (2): libgphoto2 has been compiled with the following options:
0.000151 main                        (2):  + all camlibs
0.000158 main                        (2):  + clang (C compiler used)
0.000164 main                        (2):  + ltdl (for portable loading of camlibs)
0.000169 main                        (2):  + no EXIF (for special handling of EXIF files)
0.000176 main                        (2): libgphoto2_port 0.12.0
0.000182 main                        (2): libgphoto2_port has been compiled with the following options:
0.000188 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000194 main                        (2):  + clang (C compiler used)
0.000200 main                        (2):  + ltdl (for portable loading of iolibs)
0.000206 main                        (2):  + USB (libusb1, for USB cameras)
0.000212 main                        (2):  + serial (for serial cameras)
0.000219 main                        (2):  + no resmgr (serial port access and locking)
0.000225 main                        (2):  + no ttylock (serial port locking)
0.000231 main                        (2):  + no lockdev (serial port locking)
0.000237 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000243 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000249 main                        (2): invoked with following arguments:
0.000255 main                        (2):   --capture-image-and-download
0.000262 main                        (2):   --keep-raw
0.000268 main                        (2):   --debug
0.000274 main                        (2):   --debug-logfile=logfile.txt
0.000280 main                        (2):   --filename
...
2.236468 ptp_usb_getresp             (2): Reading PTP_OC 0x1008 (Get object info) response...
2.236484 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
2.237241 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-1f 00 00 00              ....... ....    

2.237308 debug_objectinfo            (2): ObjectInfo for 'DSC_0000.NEF':
2.237326 debug_objectinfo            (2):   Object ID: 0x0b000001
2.237340 debug_objectinfo            (2):   StorageID: 0x00000000
2.237353 debug_objectinfo            (2):   ObjectFormat: 0x3000
2.237370 debug_objectinfo            (2):   ProtectionStatus: 0x0000
2.237383 debug_objectinfo            (2):   ObjectCompressedSize: 17569375
2.237396 debug_objectinfo            (2):   ThumbFormat: 0x3808
2.237413 debug_objectinfo            (2):   ThumbCompressedSize: 57600
2.237426 debug_objectinfo            (2):   ThumbPixWidth: 160
2.237439 debug_objectinfo            (2):   ThumbPixHeight: 120
2.237452 debug_objectinfo            (2):   ImagePixWidth: 4928
2.237465 debug_objectinfo            (2):   ImagePixHeight: 3280
2.237477 debug_objectinfo            (2):   ImageBitDepth: 0
2.237490 debug_objectinfo            (2):   ParentObject: 0x00000000
2.237502 debug_objectinfo            (2):   AssociationType: 0x0000
2.237515 debug_objectinfo            (2):   AssociationDesc: 0x00000000
2.237527 debug_objectinfo            (2):   SequenceNumber: 0x00000000
2.237540 debug_objectinfo            (2):   ModificationDate: 0x5c3224dd
2.237552 debug_objectinfo            (2):   CaptureDate: 0x5c3224dd
2.237563 camera_wait_for_event       (2): raw? ofc is 0x3000, name is DSC_0000.NEF
2.237584 camera_wait_for_event       (2): trying to get object size=0x10c165f
2.237594 ptp_usb_sendreq             (2): Sending PTP_OC 0x1009 (Get object) (0xb000001) request...
2.237604 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
2.237742 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 09 10-20 00 00 00 01 00 00 0b  ........ .......

2.237764 ptp_usb_getdata             (2): Reading PTP_OC 0x1009 (Get object) data...
2.237779 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
22.238592 gp_libusb1_read [libusb1.c:603](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Operation timed out (-7)
22.238672 gp_port_read [gphoto2-port.c:441](0): Reading 1024 = 0x400 bytes from port failed: Timeout reading from or writing to the port (-10)
[... cut ...]
22.240167 ptp_usb_sendreq             (2): Sending PTP_OC 0x9202 (PTP_OC_NIKON_EndLiveView) request...
22.240185 gp_port_write               (3): Writing 12 = 0xc bytes to port...
22.240428 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 02 92-21 00 00 00              ........!...    

22.240463 ptp_usb_getresp             (2): Reading PTP_OC 0x9202 (PTP_OC_NIKON_EndLiveView) response...
22.240489 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...

```
and waiting forever, can't even switch off camera.

msmeissn, 2019-01-11:
Hmm.  Can you try current libgphoto2 git.

I solved a WLAN NEF file transfers problem that happened when downloading large NEF files, by splitting into smaller partial transfers. This might also help here.