gphoto/libgphoto2 #221: Gphoto2 and libgphoto2 Version Issues

benwis, 2018-01-05:
Hi,
I'm trying to use gphoto2 and libgphoto2 and a Sony camera. I think I'm having version issues, because It correctly executes the command:
gphoto2 --capture-movie --stdout 
on my Arch Linux box desktop running photo2 2.5.15 libgphoto2 2.5.16 and libgphoto2_port 0.12.0

but it fails with this error on Debian Jessie on my Raspberry Pi:
Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
Error: Movie Capture error... Exiting.
Movie capture finished (o frames)

I also noticed another quirk:
gphoto2 --auto-detect
The camera is identified on my Arch Linux box as a Sony UM-R10C
but on my Raspberry Pi it is a Generic PTP Device

Can you provide any guidance as to which version of libgphoto2 or gphoto2 might be required to fix this?

benwis, 2018-01-05:
I'm attaching two log files. One succesfull, logfile-arch-desk.log, and one not logfile-rpi.log
[logfile-archdesk.log](https://github.com/gphoto/libgphoto2/files/1605471/logfile-archdesk.log)
[logfile-rpi.log](https://github.com/gphoto/libgphoto2/files/1605472/logfile-rpi.log)

msmeissn, 2018-01-14:
Sony Liveview capture needs libgphoto2 2.5.13 or newer. thats why it does not work on your raspberry.

closed 2018-01-14 by msmeissn