gphoto/libgphoto2 #223: Canon EOS 5DS - Silent LV Shoot

oloflarsson, 2018-01-08:
The Canon EOS 5DS has a setting called "Silent LV Shoot.". It looks like this at the back of the camera display:

[image]

Is there currently a way to control this setting from within gphoto2? If not, could that possibly be added? I'd be happy to help out by sending debug output from the Canon EOS 5DS. Just tell me what to do 🎉!

These are the config keys and values I currently get. They are from version 2.5.9 or 2.5.14, can't remember which. The point though is that I don't think I see anything about "Silent LV Shoot." in there right now 😸?

* */main/actions/syncdatetimeutc -* 0
* */main/actions/syncdatetime -* 0
* */main/actions/uilock -* 2
* */main/actions/autofocusdrive -* 0
* */main/actions/manualfocusdrive -* None
* */main/actions/cancelautofocus -* 0
* */main/actions/eoszoom -* 0
* */main/actions/eoszoomposition -* 0,0
* */main/actions/viewfinder -* 2
* */main/actions/eosremoterelease -* None
* */main/actions/opcode -* 0x1001,0xparam1,0xparam2
* */main/settings/datetimeutc -* 1503991205
* */main/settings/datetime -* 1503987605
* */main/settings/reviewtime -* 2 seconds
* */main/settings/output -* Off
* */main/settings/movierecordtarget -* None
* */main/settings/evfmode -* 1
* */main/settings/ownername -* 
* */main/settings/artist -* 
* */main/settings/copyright -* asdfasdf
* */main/settings/customfuncex -* 15c,1,4,150,6,70c,40,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,fe,ff,0,3,70,30,0,0,0,0,0,0,0,0,0,0,0,14,50,0,0,0,0,7,3,0,ff,ff,ff,0,ff,ff,ff,f,706,1,0,70f,2,0,2,80e,1,0,813,1,0,814,1,0,
* */main/settings/focusinfo -* eosversion=2,size=8688x5792,size2=8688x5792,points={{-435,652,283,283},{-966,652,283,283},{-1425,652,283,283},{-1860,652,283,283},{-2295,652,283,283},{435,978,283,283},{0,978,283,283},{-435,978,283,283},{-1860,326,283,283},{-2295,326,283,283},{2295,652,283,283},{1860,652,283,283},{1425,652,283,283},{966,652,283,283},{435,652,283,283},{0,652,283,283},{1860,326,283,283},{1425,326,283,283},{966,326,283,283},{435,326,283,283},{0,326,283,283},{-435,326,283,283},{-966,326,283,283},{-1425,326,283,283},{435,0,283,283},{0,0,283,283},{-435,0,283,283},{-966,0,283,283},{-1425,0,283,283},{-1860,0,283,283},{-2295,0,283,283},{2295,326,283,283},{-966,-326,283,283},{-1425,-326,283,283},{-1860,-326,283,283},{-2295,-326,283,283},{2295,0,283,283},{1860,0,283,283},{1425,0,283,283},{966,0,283,283},{-2295,-652,283,283},{2295,-326,283,283},{1860,-326,283,283},{1425,-326,283,283},{966,-326,283,283},{435,-326,283,283},{0,-326,283,283},{-435,-326,283,283},{1425,-652,283,283},{966,-652,283,283},{435,-652,283,283},{0,-652,283,283},{-435,-652,283,283},{-966,-652,283,283},{-1425,-652,283,283},{-1860,-652,283,283},{435,-978,283,283},{0,-978,283,283},{-435,-978,283,283},{2295,-652,283,283},{1860,-652,283,283}},select={},unknown={00000002000000000000000000000000ffff}
* */main/settings/focusarea -* 134233919
* */main/settings/autopoweroff -* 240
* */main/settings/depthoffield -* 0
* */main/settings/capturetarget -* Internal RAM
* */main/settings/capture -* 0
* */main/status/serialnumber -* 8ff9fd4117f045b8a1a4d5da70daef5b
* */main/status/manufacturer -* Canon Inc.
* */main/status/cameramodel -* Canon EOS 5DS
* */main/status/deviceversion -* 3-1.0.1
* */main/status/vendorextension -* None
* */main/status/model -* 2147484546
* */main/status/ptpversion -* 256
* */main/status/Battery -* el", "66%
* */main/status/batterylevel -* 75%
* */main/status/lensname -* EF50mm f/1.2L USM
* */main/status/eosserialnumber -* 033021001133
* */main/status/availableshots -* 4682
* */main/imgsettings/imageformat -* RAW
* */main/imgsettings/imageformatsd -* RAW
* */main/imgsettings/imageformatcf -* RAW
* */main/imgsettings/iso -* 100
* */main/imgsettings/whitebalance -* Auto
* */main/imgsettings/colortemperature -* 3900
* */main/imgsettings/whitebalanceadjusta -* 0
* */main/imgsettings/whitebalanceadjustb -* 0
* */main/imgsettings/whitebalancexa -* 0
* */main/imgsettings/whitebalancexb -* 0
* */main/imgsettings/colorspace -* AdobeRGB
* */main/capturesettings/focusmode -* One Shot
* */main/capturesettings/continuousaf -* Off
* */main/capturesettings/autoexposuremode -* Manual
* */main/capturesettings/drivemode -* Single
* */main/capturesettings/picturestyle -* Neutral
* */main/capturesettings/aperture -* 11
* */main/capturesettings/shutterspeed -* 1/200
* */main/capturesettings/meteringmode -* Evaluative
* */main/capturesettings/bracketmode -* 0
* */main/capturesettings/aeb -* off
* */main/other/d049 -* -2147482750
* */main/other/d402 -* Canon EOS 5DS
* */main/other/d407 -* 1
* */main/other/d406 -* Unknown Initiator
* */main/other/d303 -* 1
* */main/other/5001 -* 66

msmeissn, 2018-01-15:
can you capture debug output of list-config for the 3 modes select in this menu?

gphoto2 --list-config --debug --debug-logfile=mode1.log
... change mode on camrea ... 
gphoto2 --list-config --debug --debug-logfile=mode2.log
... cjhange mode to camera
gphoto2 --list-config --debug --debug-logfile=mode3.log

and attach them here

oloflarsson, 2018-01-15:
Mode 1: https://pastebin.com/raw/pjtcNfrN
Mode 2: https://pastebin.com/raw/6iNSVdPp
Disable: https://pastebin.com/raw/WkQK6Kh8

msmeissn, 2019-03-03:
forgot to follow up here so far ... 

The changing property values do not seem to reflect to silent lv shooting.
- the time reported by the camera changes
- the alomode changes between mode1 and mode2
- some values called "exposure simulation mode" changes between 1 and 2.