gphoto/libgphoto2 #224: Canon EOS 5DS - Missing /main/status/shuttercounter

oloflarsson, 2018-01-08:
The `Canon EOS 5D Mark III` has a a config key `/main/status/shuttercounter` that's really useful for me. I look at it to estimate the camera wear and remaining life span.

After changing to use of `Canon EOS 5DS` this config key is no longer available. I kinda miss it to be honest 😢.

Could this config key be added for `Canon EOS 5DS`? Or is it deprecated and there is now a new better method to check the camera wear?

Just like for this ticket https://github.com/gphoto/libgphoto2/issues/223 I would be happy to send any debug output you might need from the camera.

Also thank's a lot for you work on gphoto2. Excellent software!

oloflarsson, 2018-01-10:
Is this ticket perhaps a duplicate of: https://github.com/gphoto/gphoto2/issues/81

oloflarsson, 2018-01-15:
Of course. I ran it with `gphoto 2.5.9` and `Canon EOS 5DS`:
https://pastebin.com/raw/NCqVw3AZ

joelcomp1, 2019-01-29:
anyone find a work around to this? I find this very useful as well.

ngortheone, 2020-04-13:
@msmeissn I also have this problem on my 5D Mark IV.
I am willing to provide any additional information necessary to assist you!

msmeissn, 2020-04-13:
Canon is not exporting the shuttercount value on all EOS series for some reason.

ngortheone, 2020-04-13:
Is there a different way of getting it? (including outside of capabilities of gphoto?) I found a few paid apps and I think of buying one, but I am worried that it might be fraud/fake.
What do you think?

Below the line:
------------------
I have fount a property named "availableshots" but it does not increase as I take more pictures. (seems to me set at 108435), it is likely something different...

msmeissn, 2020-04-29:
availableshots is the number of shots possible for the current "free space" on the Flash / SD Card.