gphoto/libgphoto2 #226: error: 'aox.la' is not a valid libtool archive

Bobrow, 2018-01-13:
I am trying to install this on a raspberry pi and when I do `sudo make install` it runs for a bit everything working then it gets to "aox.la" and says it isn't a valid archive.

Thanks in advance for help

PS: This is the output from running sudo make install:
`Making install in libgphoto2_port
make[1]: Entering directory '/home/pi/libgphoto2/libgphoto2_port'
Making install in po
make[2]: Entering directory '/home/pi/libgphoto2/libgphoto2_port/po'
/bin/sh ../auto-aux/mkinstalldirs /usr/local/share
installing cs.gmo as /usr/local/share/locale/cs/LC_MESSAGES/libgphoto2_port-12.mo
installing da.gmo as /usr/local/share/locale/da/LC_MESSAGES/libgphoto2_port-12.mo
installing de.gmo as /usr/local/share/locale/de/LC_MESSAGES/libgphoto2_port-12.mo
installing es.gmo as /usr/local/share/locale/es/LC_MESSAGES/libgphoto2_port-12.mo
installing eu.gmo as /usr/local/share/locale/eu/LC_MESSAGES/libgphoto2_port-12.mo
installing fi.gmo as /usr/local/share/locale/fi/LC_MESSAGES/libgphoto2_port-12.mo
installing fr.gmo as /usr/local/share/locale/fr/LC_MESSAGES/libgphoto2_port-12.mo
installing it.gmo as /usr/local/share/locale/it/LC_MESSAGES/libgphoto2_port-12.mo
installing ja.gmo as /usr/local/share/locale/ja/LC_MESSAGES/libgphoto2_port-12.mo
installing nl.gmo as /usr/local/share/locale/nl/LC_MESSAGES/libgphoto2_port-12.mo
installing pl.gmo as /usr/local/share/locale/pl/LC_MESSAGES/libgphoto2_port-12.mo
installing pt_BR.gmo as /usr/local/share/locale/pt_BR/LC_MESSAGES/libgphoto2_port-12.mo
installing ru.gmo as /usr/local/share/locale/ru/LC_MESSAGES/libgphoto2_port-12.mo
installing sk.gmo as /usr/local/share/locale/sk/LC_MESSAGES/libgphoto2_port-12.mo
installing sr.gmo as /usr/local/share/locale/sr/LC_MESSAGES/libgphoto2_port-12.mo
installing sv.gmo as /usr/local/share/locale/sv/LC_MESSAGES/libgphoto2_port-12.mo
installing uk.gmo as /usr/local/share/locale/uk/LC_MESSAGES/libgphoto2_port-12.mo
installing vi.gmo as /usr/local/share/locale/vi/LC_MESSAGES/libgphoto2_port-12.mo
installing zh_CN.gmo as /usr/local/share/locale/zh_CN/LC_MESSAGES/libgphoto2_port-12.mo
installing zh_TW.gmo as /usr/local/share/locale/zh_TW/LC_MESSAGES/libgphoto2_port-12.mo
if test "libgphoto2_port" = "gettext-tools"; then \
  /bin/sh ../auto-aux/mkinstalldirs /usr/local/share/gettext/po; \
  for file in Makefile.in.in remove-potcdate.sin quot.sed boldquot.sed en@quot.header en@boldquot.header insert-header.sin Rules-quot   Makevars.template; do \
    /usr/bin/install -c -m 644 ./$file \
		    /usr/local/share/gettext/po/$file; \
  done; \
  for file in Makevars; do \
    rm -f /usr/local/share/gettext/po/$file; \
  done; \
else \
  : ; \
fi
make[2]: Leaving directory '/home/pi/libgphoto2/libgphoto2_port/po'
Making install in libgphoto2_port
make[2]: Entering directory '/home/pi/libgphoto2/libgphoto2_port/libgphoto2_port'
make[3]: Entering directory '/home/pi/libgphoto2/libgphoto2_port/libgphoto2_port'
 /bin/mkdir -p '/usr/local/lib'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libgphoto2_port.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libgphoto2_port.so.12.0.0 /usr/local/lib/libgphoto2_port.so.12.0.0
libtool: install: (cd /usr/local/lib && { ln -s -f libgphoto2_port.so.12.0.0 libgphoto2_port.so.12 || { rm -f libgphoto2_port.so.12 && ln -s libgphoto2_port.so.12.0.0 libgphoto2_port.so.12; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libgphoto2_port.so.12.0.0 libgphoto2_port.so || { rm -f libgphoto2_port.so && ln -s libgphoto2_port.so.12.0.0 libgphoto2_port.so; }; })
libtool: install: /usr/bin/install -c .libs/libgphoto2_port.lai /usr/local/lib/libgphoto2_port.la
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
[... 166 lines cut ...]
libtool: relink:  gcc -shared  -fPIC -DPIC  adc65/.libs/adc65.o   -L/usr/local/lib -lgphoto2 -lgphoto2_port  -g -O2   -Wl,-soname -Wl,adc65.so -Wl,-version-script -Wl,.libs/adc65.ver -o .libs/adc65.so
libtool: install: /usr/bin/install -c .libs/adc65.soT /usr/local/lib/libgphoto2/2.5.16.1/adc65.so
libtool: install: /usr/bin/install -c .libs/adc65.lai /usr/local/lib/libgphoto2/2.5.16.1/adc65.la
libtool: warning: relinking 'agfa_cl20.la'
libtool: install: (cd /home/pi/libgphoto2/camlibs; /bin/bash "/home/pi/libgphoto2/libtool"  --tag CC --mode=relink gcc -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -module -no-undefined -avoid-version -export-dynamic -export-symbols ../camlibs/camlib.sym -rpath /usr/local/lib/libgphoto2/2.5.16.1 -o agfa_cl20.la agfa-cl20/agfa_cl20.lo ../libgphoto2/libgphoto2.la ../libgphoto2_port/libgphoto2_port/libgphoto2_port.la )
libtool: relink: echo "{ global:" > .libs/agfa_cl20.ver
libtool: relink:  cat ../camlibs/camlib.sym | sed -e "s/\(.*\)/\1;/" >> .libs/agfa_cl20.ver
libtool: relink:  echo "local: *; };" >> .libs/agfa_cl20.ver
libtool: relink:  gcc -shared  -fPIC -DPIC  agfa-cl20/.libs/agfa_cl20.o   -L/usr/local/lib -lgphoto2 -lgphoto2_port  -g -O2   -Wl,-soname -Wl,agfa_cl20.so -Wl,-version-script -Wl,.libs/agfa_cl20.ver -o .libs/agfa_cl20.so
libtool: install: /usr/bin/install -c .libs/agfa_cl20.soT /usr/local/lib/libgphoto2/2.5.16.1/agfa_cl20.so
libtool: install: /usr/bin/install -c .libs/agfa_cl20.lai /usr/local/lib/libgphoto2/2.5.16.1/agfa_cl20.la
Usage: /home/pi/libgphoto2/libtool [OPTION]... [MODE-ARG]...
Try 'libtool --help' for more information.
libtool:   error: 'aox.la' is not a valid libtool archive
Makefile:1748: recipe for target 'install-camlibLTLIBRARIES' failed
make[3]: *** [install-camlibLTLIBRARIES] Error 1
make[3]: Leaving directory '/home/pi/libgphoto2/camlibs'
Makefile:3587: recipe for target 'install-am' failed
make[2]: *** [install-am] Error 2
make[2]: Leaving directory '/home/pi/libgphoto2/camlibs'
Makefile:3426: recipe for target 'install-recursive' failed
make[1]: *** [install-recursive] Error 1
make[1]: Leaving directory '/home/pi/libgphoto2/camlibs'
Makefile:689: recipe for target 'install-recursive' failed
make: *** [install-recursive] Error 1`

msmeissn, 2018-01-13:
can you make clean and try to bnuild again. it might just be a hickup

Bobrow, 2018-01-18:
I got it to work, because I had the github version and another version I just swapped the .la files between them.

closed 2018-01-18 by Bobrow

msmeissn, 2018-01-19:
good to hear! libtool is still also blck magic for me, which mostly works :/