gphoto/libgphoto2 #227: Sony a5100 long delay before capture

mkunzBGC, 2018-01-16:
Hi,
I am using gphoto2 2.5.15 with libgphoto2 2.5.16.
`gphoto2 --capture-image-and-download`
works, but there is a ~7 seconds delay between sending the command and the actual shot.
On the other hand, when using the gphoto shell
`gphoto2 --shell`
it takes ~5 seconds to enter the shell, but then
`capture-image-and-download`
takes a shot within less than 1 second.

I would like to control the camera from a script, so shell mode is not what I need. Is there any way to reduce the delay in command mode?

Furthermore, while some parameters (e.g. f-number) can be controlled both from the shell or a gphoto command, others such as ISO work only from shell mode.
`gphoto2 --set-config ISO=100`
does not produce output and does not change the ISO, but it breaks the communication with the camera. Further commands return errors:
```
$gphoto2 --summary
                                                                               
*** Error ***             
PTP I/O Error

*** Error ***              
An error occurred in the io-library ('I/O problem'): No error description available
*** Error (-7: 'I/O problem') ***
```
Control can only be regained after turning the camera off and on again.

### Detailed info
`gphoto2 -v`: [gphotoVersion.log](https://github.com/gphoto/libgphoto2/files/1634515/gphotoVersion.log)
`gphoto2 --list-all-config --debug --debug-logfile=allconfig.log`: [allconfig.log](https://github.com/gphoto/libgphoto2/files/1634502/allconfig.log)
`gphoto2 --set-config F-Number=8 --debug --debug-logfile=setFNumber.log`: [setFNumber.log](https://github.com/gphoto/libgphoto2/files/1634527/setFNumber.log)
`gphoto2 --set-config ISO=400 --debug --debug-logfile=setISO.log`: [setISO.log](https://github.com/gphoto/libgphoto2/files/1634532/setISO.log)
And then, after trying unsuccessfully to set ISO:
`gphoto2 --set-config F-Number=4 --debug --debug-logfile=setFNumberAfterIsO.log`: [setFNumberAfterIsO.log](https://github.com/gphoto/libgphoto2/files/1634541/setFNumberAfterIsO.log)
with output as above:
```                                                                               
*** Error ***              
PTP I/O Error

*** Error ***              
An error occurred in the io-library ('I/O problem'): No error description available

*** Error ***              
PTP I/O Error

*** Error ***              
An error occurred in the io-library ('I/O problem'): No error description available
*** Error (-7: 'I/O problem') ***      
```

msmeissn, 2018-03-30:
can you also add a trace for the long capture startup?  this should not be the case

msmeissn, 2018-03-30:
The iso issue can be solved by wiriting "iso" in lowercase. (There is a uppercase ISO entry, but this uses PTP generic methods which do not work on the Sony reliably.)

I fixed the bug that made the camera unresponsive, but it will likely not work with uppercase ISO.

msmeissn, 2018-03-30:
same if F-Number does not work, switch to "f-number".

mkunzBGC, 2018-04-26:
Finally I have time to come back to this. I have just built libgphoto2 2.5.17.1 and gphoto2 2.5.17.1.
Indeed, the crash when trying to change the uppercase 'ISO' is fixed. As you expected, to actually change ISO it is required to write lower case 'iso' in the --set-config command. In contrast, both "F-Number" and "f-number" work equally well.

Good news also about the long delay: using the new version,
`gphoto2 --capture-image-and-download`
takes an image after ~1s of delay. I generated the debug output anyways:
'gphoto2 --capture-image-and-download --debug --debug-logfile=capture.log':  [capture.log](https://github.com/gphoto/libgphoto2/files/1951385/capture.log)

Thank you for your great work!

msmeissn, 2018-04-26:
you are welcome! :)

closed 2018-04-26 by msmeissn