gphoto/libgphoto2 #228: Pentax (k5) image reading very slow

mccodesurfer, 2018-01-16:
Takes about 30 secs to read an image. Tolerable only when being used for interval shooting.

mccodesurfer, 2018-02-03:
Is there something I can do to fix this???

mccodesurfer, 2018-02-03:
I'm looking at the code and found this in pslr.c:

#define BLKSZ 65536 /* Block size for downloads; if too big, we get
                     * memory allocation error from sg driver */

This is pretty small and probably causing the delay since the ipslr_download has to retry thousands of times for a 20 MB image download. maybe this sg driver error is no longer valid. can the block size be increased?

msmeissn, 2018-02-03:
30 seconds sounds very loing either way.

you could try bumping this value and see if it helps.

also a gphoto2 debuglog will have timestamps in front, to see if there is soemthing else blocking.

msmeissn, 2018-02-03:
so if the value increment does not help, run with --debug --debug-logfile=xx.log  attached to the gphoto2 commandline and look for bigger timestamp increments

mccodesurfer, 2018-02-04:
Thanks. I'm trying to work on it but I'm dead in the water with a missing file in gphoto2: https://github.com/gphoto/gphoto2/issues/115

mccodesurfer, 2018-02-05:
well, it looks more like 4 seconds to retreive the image file from the camera. I guess the problem is somewhere else. StellarMate may be in debug mode and converting the RAW to FITS.

mccodesurfer, 2018-02-05:
There seems to be a number of factors in StellarMate - like FITS conversion. I'll try a larger block size to see if there is significant speed increase. Otherwise, I'll pursue StellarMate OS for this speed issue.

mccodesurfer, 2018-02-06:
So the 4 seconds is actually the camera making the file available. The image retrieve is a fraction of a second. I'm seeing other delays like wifi transfer delay and dng2fits conversion. Looks like there's no issue with gphoto here.

Thank you @msmeissn for you help!

closed 2018-02-08 by mccodesurfer