gphoto/libgphoto2 #229: Olympus E-M1 MarkII not supported?

haysclark, 2018-01-21:
It would be awesome to add full support for the Olympus E-M1, including live capture, etc. Olympus has a nice program that can be used with USB tethered, additionally, programs like DragonFrame also have support for Live Capture, etc.  The E-M1 does support PTP unlike many other Olympus models, so if there is anything I can do to help aid in the process please let me know how I can be of service, or where I could start.

msmeissn, 2018-01-21:
can you report output of:
lsusb
gphoto2 --summary 
gphoto2 --debug --debug-logfile=list-config.log --list-all-config
gphoto2 --capture-image

(also attacg list-config.log )

haysclark, 2018-01-24:
@msmeissn I have put all of this information into a Gist.  The only quirk is that the 'lsusb' information is from my Linux box.  However, gphoto2 could not communicate with the camera on my linux setup (I need to tinker to see what is up)

I was able to collect all of the other information from OSX.  Hopefully, that is not an issue.

```gphoto2 --capture-image``` is the only command that does not work.  I included both the output and the output from ```env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image ```.

Here is the Gist:
https://gist.github.com/haysclark/cb100282f78d6f2eda99cf37fa639c65

Please let me know how and if I can help.

haysclark, 2018-01-25:
@msmeissn I ran the commands above on my Linux box running the latest version of libgphoto2 and gphoto2. The files are all located here in my [gist](https://gist.github.com/haysclark/ccee7245081b4b94851bd0f56d65861b).

```gphoto2 --capture-image``` is the only command that errored. I included the debug output in the Gist.
```
*** Error ***              
Sorry, your camera does not support generic capture list-config.log gphoto2-debug
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***
```

File name key:

| command | file name |
|--|:--:|
| lsusb | lsusb.log |
| gphoto2 --summary | gphoto2-summary.log |
| gphoto2 --debug --debug-logfile=list-config.log --list-all-config | gphoto2-debug.log |
| list-config.log | list-config.log |
| gphoto2 --capture-image | gphoto2-capture-image.log |
| env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image | my-logfile.txt |

haysclark, 2018-01-25:
@msmeissn I have started to dig into DTrace with two OSX programs that can control the E-M1.  I am new to DTrace, so progress is a bit slow, but I am trying to understand what event they are using to trigger the capture.  Is there an easy way to fire random events to the camera and see what happens, via a CLI perhaps?

msmeissn, 2018-01-26:
we need some form of USB sniffing.

The summar/debug shows quite a number of properties that seem related to camra control (look for the ISO steps lists for instance).

But there are not many vendor PTP opcodes, most are in the MTP range, only 1 is outside of it.

Firiing random PTP opcodes at a camera is risky though, but could be with   gphoto2 --set-config opcode=0x........ command

haysclark, 2018-02-15:
@msmeissn Do you recommend a specific USB sniffing app or hardware? I have access to Linux/Mac/PC.  I could also continue to use DTrace on my Mac.

Could you clarify the process I should take for USB sniffing?  For instance, should I set the camera ISO 200, start sniffing USB traffic, then change the ISO to 300 (via software), stop the capture, then repeat the process for ISO 300 -> ISO 400, repeat? 

Another developer I reached out too suggested doing the same steps above, running ```gphoto2 --summary``` before and after a setting change, then performing a DIFF.

What would you recommend?

elijahparker, 2018-02-16:
@haysclark -- sorry for my absence on the thread! I've spent the last couple weeks with some rental cameras and have barely kept up with everything going on! Anyway, I just upgraded to the Beagle USB 480, a really nice hardware USB sniffer, and I am selling my Beagle USB 12 on eBay right now, but I could loan it to you first if you're interested. Otherwise you can also send me your camera and I can do a bunch of captures for a day and send it back. Let me know what works best for you. 

Once we've got the protocol captured it should be fairly quick to get it all working. I just got the GH5 working here this week.

elijahparker, 2018-02-26:
An update here -- I just sold my Beagle USB 12, but I instead scheduled a rental to test with that's arriving tomorrow.  I should have some info to help here soon.

haysclark, 2018-03-02:
@elijahparker So sorry I missed you, I was unplugged all last week doing drywall repairs.  Can I help you out with the rental charges?

HelgeSuess, 2018-04-03:
Hi!
I just did a few recordings with a software USB sniffer. The E-M1 ii is pretty noisy on USB when connected to Olympus Capture. There's masses of traffic when the camera is idle.
I did a trace when teh camera is connected to USB and one for pressing the release buttton on the camera and on the PC.
[EM1ii_logs.zip](https://github.com/gphoto/libgphoto2/files/1873262/EM1ii_logs.zip)

haysclark, 2018-05-31:
I have been syncing up with @elijahparker and he has been plugging away on the E-M1 Mark II and some other Olympus models.  While he was successful at getting the E-M5 Mark II running, the E-M1 Mark II seems to be much pickier for the init sequence.  The additional challenge is that the cameras do not follow PTP specs.

msmeissn, 2018-07-21:
the e-m5 mark2 code is merged back to libgphoto2 git now

Entropy512, 2018-08-30:
FYI, if you are sniffing USB between a Windows application and a device (as opposed to between a "custom" device such as a gaming console and one of its peripherals) - You can sniff it using Wireshark if you run Windows in a VM.  https://wiki.wireshark.org/CaptureSetup/USB

I will be capturing traces of Sony's Imaging Edge talking to an A7III in the next month or two, but I've changed laptops and need to set up a VM again...

BeebBenjamin, 2019-05-07:
Hi, I tried out gphoto2 2.5.20 with libgphoto2 2.5.22 installed with an Olympus E-M1 Mark II and could not get it to complete the capture process.  It hangs and you have to force close gphoto2.  In addition the Olympus switches to a state where it cannot be turned off without removing the battery.

One thing I noticed is that when you use --auto-detect it lists the camera as "Olympus E-M5 Mark II "

berylius, 2019-05-17:
Hi,

do you have libusb1 installed? see #386

elijahparker, 2019-05-17:
The M1 II and M5 II share the same USB  ID, but they also seem to be
equivalent via USB.

Libusb 0.1 does NOT work with Olympus.

BeebBenjamin, 2019-05-23:
I have a rental model OE-M1 MKII to play around with.  Any suggestions for what to try?  Capture just freezes on a Raspberry Pi, and on Ubuntu there is a an error that says "Error initialising camera: -60: Could  not lock the device".

BeebBenjamin, 2019-06-13:
I have to return this camera soon.  I have tried to see what the issue is, but without help I have been unable to trace the problem.  Which version of libusb do I need to make sure I have installed?  And how do I build libgphoto2 to use a version that does work with an Olympus?

berylius, 2019-06-13:
You can use https://github.com/gonzalo/gphoto2-updater ton install the latest version. 
And then be sure to have libusb1 installed, the script installs libusb-dev you can remove it and install libusb-1.0-0-dev instead.

BeebBenjamin, 2019-06-13:
Thanks.  I just changed the script to install that instead of libusb-dev.  I had both versions installed, which might be what is causing the problem.  It will definitely be why the Raspberry Pi isn't working, as I use gphoto2-updater to update this.  I will confirm if this fixes the issue on Saturday.  Thanks for the help!

cparg, 2019-08-03:
I have the same problem. 
My new OM-D E-M5 II captures only the first time after switching on the camera.
I actually have latest (git) version of libgphoto and gphoto2 now running in the debugger but I am lost what to look for. Please advise. 
It would be also possible to use the Olympus Capture tool in a Windows VM and sniff the USB traffic. But I think that has already been done w/o much success.

elijahparker, 2019-08-05:
One thing to remember with OM-D is that the init sequence resets the USB
connection in a way that doesn't seem to work with libusb0.1, so make sure
you're using libusb1.0.  For me, despite compiling with libusb1.0, it was
still using the old one until I change the name of the shared library
itself so that libusb0.1 couldn't be found (you could also delete it).

cparg, 2019-08-05:
Hello Elijah,

I did remove the libusb-0.1 version and tried again, but the bahaviour
is the same.
First two captures work, the third (and any later) fails  :(

    cparg@crusty : ~/tmp$ ll /usr/lib64/libusb* /usr/lib64/off.lib*
    -rwxr-xr-x. 1 root root 103680 11. Apr 2018 
    /usr/lib64/libusb-1.0.so.0.1.0
    lrwxrwxrwx. 1 root root     19 20. Sep 2017 
    /usr/lib64/libusbmuxd.so.4 -> libusbmuxd.so.4.0.0
    -rwxr-xr-x. 1 root root  32680  6. Aug 2017 
    /usr/lib64/libusbmuxd.so.4.0.0
    lrwxrwxrwx. 1 root root     24 14. Mai 2018 
    /usr/lib64/libusbredirhost.so.1 -> libusbredirhost.so.1.0.0
    -rwxr-xr-x. 1 root root  41008 11. Apr 2018 
    /usr/lib64/libusbredirhost.so.1.0.0
    lrwxrwxrwx. 1 root root     26 14. Mai 2018 
    /usr/lib64/libusbredirparser.so.1 -> libusbredirparser.so.1.0.0
    -rwxr-xr-x. 1 root root  31904 11. Apr 2018 
    /usr/lib64/libusbredirparser.so.1.0.0
    lrwxrwxrwx. 1 root root     19  5. Aug 23:10 /usr/lib64/libusb.so ->
    libusb-1.0.so.0.1.0
    -rwxr-xr-x. 1 root root  23936 10. Jun 2014 
    /usr/lib64/off.libusb-0.1.so.4.4.4

Regards,

    Christian

Kelvin-Wu, 2020-01-12:
Hi all, I see E-M1 markII can use libgphoto2 2.5.19 or newer to support capture function. Did anyone succeed in it?
[gphoto](http://www.gphoto.org/doc/remote/)

mad-marco, 2020-05-11:
Has anyone managed to get the E-M1 II working fully in gphoto2? I opened an incident a couple of weeks ago in the gphoto2 before I found this issue posted here. I'm having similar issues to other people with the E-M1 II. It's almost working, but just not quite.

gphoto2 --capture-image
ERROR: Could not capture image.
ERROR: Could not capture.

Note that the camera HAS successfully recorded an image to the internal SD card.

Further testing showed the following:
gphoto2 --capture-preview - I get a capture_preview.jpg image transferred to the RPI
gphoto2 --capture-movie 2 - I get 3 frames captured to movie.mjpg on the RPI
gphoto2 --list-files - Returns the following:
There is no file in folder '/'.

Note that this does not list the files or folders that are on the SD Card. If I switch the camera to Storage mode, then I can see the files and folders.

gphoto2 --storage-info - Does not return anything when in PC capture mode, but correctly shows the SD Card when the camera is switched to Storage mode.

I am thinking that the inability to connect to the SD card on the camera when in PC connect mode is the real issue or at least related. Anyone got any thoughts on this?

If there is anything that I can do to assist in resolving this issue, then please let me know. I can do any testing that would be required.

msmeissn, 2020-05-23:
we have tracked this in the other Olympus E-1 Mark 2 issue https://github.com/gphoto/gphoto2/issues/310

closed 2020-05-23 by msmeissn