gphoto/libgphoto2 #230: Trying to use with Sony A7RIII ... almost works

sipanov, 2018-01-22:
I was hoping I'd be able to export and manage videos and photos in my new camera without having to moving SD card from camera to computer and back. It almost works, but it looks like I can not delete images and videos using MTP or PTP  mode (and I am not comfortable using Mass Storage mode for deletion - I did not even try it because there seems to be some DB indexing files).

  here is what I have so far:
 Camera have different USB IDs in different modes:

- [Off(Charger)] 054c:0994
- [ Auto] 054c:0bfe
- [ Mass Storage] 054c:0bff 
- [MTP] 054c:0c00
- [PC Remote] 054c:0c33 

Before I've added rules for _udev_ for and installed _exfat-utils_ and _fuse-exfat_ none of the camera USB mode worked in any way.  Installation of the  _exfat-utils_ and _fuse-exfat_  fixed "Mass Storage" and "Auto"(which defaults/switched to "Mass Storage") modes.
To fix _udev_ rules I've created `/etc/udev/rules.d/41-sony-a7riii.rules` file with
```
ATTRS{idVendor}=="054c", ATTRS{idProduct}=="0bfe", ENV{ID_GPHOTO2}="1", ENV{GPHOTO2_DRIVER}="proprietary"
ATTRS{idVendor}=="054c", ATTRS{idProduct}=="0c00", ENV{ID_GPHOTO2}="1", ENV{GPHOTO2_DRIVER}="proprietary"
ATTRS{idVendor}=="054c", ATTRS{idProduct}=="0c33", ENV{ID_GPHOTO2}="1", ENV{GPHOTO2_DRIVER}="proprietary"
```
fixed "MTP" and ""PC Remote"(?PTP?) modes ("Auto" mode now defaults to "Mass Storage" now and _gphoto2_ does not even see camera now in the "Auto" mode which drops from _0bfe_  to _0bff_ right away). In both "MTP" and "PC Remote" modes I get the same set of capabilities
```
$ gphoto2 -a
Abilities for camera             : USB PTP Class Camera                        
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Image
                                 : Preview
Configuration support            : yes
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes
```
but in the "MTP" mode the _capture_ feature does not work and in the "PC Remote" mode the _capture_ feature works but folders and files can not be listed (by _gphoto2_ and _shotwell_). The _capture_ feature is not some thing I care about, I was using it as a test. The "MTP" mode would be just fine, but I can not delete any files in file manager, from the command line, in the _shotwell_, or directly by _gphoto2_ utility. 

[gphoto2_cmds_ms_mode.txt](https://github.com/gphoto/libgphoto2/files/1650503/gphoto2_cmds_ms_mode.txt)
[gphoto2_cmds_mtp_mode.txt](https://github.com/gphoto/libgphoto2/files/1650504/gphoto2_cmds_mtp_mode.txt)
[gphoto2_cmds_pcctl_mode.txt](https://github.com/gphoto/libgphoto2/files/1650505/gphoto2_cmds_pcctl_mode.txt)
[logfile_ciad_mtp.txt](https://github.com/gphoto/libgphoto2/files/1650506/logfile_ciad_mtp.txt)
[summary_mtp.txt](https://github.com/gphoto/libgphoto2/files/1650507/summary_mtp.txt)
[summary_pcctl.txt](https://github.com/gphoto/libgphoto2/files/1650508/summary_pcctl.txt)
[system_log_ms_mode.txt](https://github.com/gphoto/libgphoto2/files/1650509/system_log_ms_mode.txt)
[system_log_mtp_mode.txt](https://github.com/gphoto/libgphoto2/files/1650510/system_log_mtp_mode.txt)
[system_log_mtp_pcctl.txt](https://github.com/gphoto/libgphoto2/files/1650511/system_log_mtp_pcctl.txt)
[usb_ids.txt](https://github.com/gphoto/libgphoto2/files/1650512/usb_ids.txt)

Entropy512, 2018-01-24:
I believe that most of the issues you are describing are common to all Sony cameras.

Is Sony's own software capable of capture and file management in the same mode?

I use Mass Storage mode/card transfer for deletion on my A6500 - you have to either force a database rebuild or format the card, I usually just do the latter.  Sony's image database crap is kinda annoying...

msmeissn, 2018-01-25:
this is what I also see.. PC Control mode does allow capture, but no file access.
MTP has file access, but no capture.

this is a limitation of the Sony alpha firmware I fear.

closed 2018-02-11 by msmeissn

gabrielstuff, 2019-04-17:
I was poking around and readin about the sony alpha. Reading through http://markogronroos.blogspot.com/2018/02/using-sony-6500-in-linux.html makes me think that the Sony Alpha now support capture and download. Or did I miss something ?

Thanks !

msmeissn, 2019-05-05:
yes, Sony Alphas are supported for capture. (but better check the lists ;) The 6500 is

bombomus63740, 2023-12-07:
Can you switch from "MTP" to "PC Control"  mode from the remote computer , without touching the camera?

msmeissn, 2023-12-08:
not to our current knowledge.