gphoto/libgphoto2 #234: libusb_submit_transfer ... Entity not found (-5)

pwwong, 2018-02-07:
Hi All,
Using macOS Sierra and gphoto2 2.5.15, libgphoto2 2.5.16, when trying to open my camera (which used to work before I had to reload by MacBook), I get the following from the debug log screen:

0.089642 gp_port_open                (2): Opening USB port...
0.089648 gp_libusb1_open             (2): ()
0.089898 gp_libusb1_open             (2): claiming interface 0
0.090484 gp_libusb1_queue_interrupt_urbs [libusb1.c:706](0): 'libusb_submit_transfer (port->pl->transfers[i])' failed: Entity not found (-5)
0.090539 gp_context_error            (0): An error occurred in the io-library ('I/O problem'): No error description available

*** Error ***              
An error occurred in the io-library ('I/O problem'): No error description available
*** Error (-7: 'I/O problem') ***       

0.090570 gp_camera_free              (2): Freeing camera...

Any ideas?

Thanks

msmeissn, 2018-02-10:
can you attach the full debuglogfile?

pwwong, 2018-02-20:
Sorry for the delay. The email ended up in the spam folder.

Anyway, how do you want the debuglogfile generated? What options?

Thanks
-phil

msmeissn, 2018-02-20:
the full debugging output fvrom the window you pasted the first entries... but the above has some parts already, it is weird that this happens on open.

Is PTPCamera stopped as necessary for MacOS?

What camera is it?

pwwong, 2018-02-20:
Attached is the logfile.

Never had to stop PTPCamera before. There was some update either to 
gphoto2 or MacOS that caused gphoto2 to stop working. I can't tell which.

The Camera is an Olympus C-3000

-- 

"Pretending that the all-but-inevitable is inconceivable doesn't
make it impossible."

Phillip W. Wong, PE (phil@weballey.com)

0.000204 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000240 main                        (2): gphoto2 2.5.15
0.000249 main                        (2): gphoto2 has been compiled with the following options:
0.000254 main                        (2):  + /usr/bin/clang (C compiler used)
0.000259 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000263 main                        (2):  + exif (for displaying EXIF information)
0.000267 main                        (2):  + no cdk (for accessing configuration options)
0.000271 main                        (2):  + no aa (for displaying live previews)
0.000275 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000279 main                        (2):  + readline (for easy navigation in the shell)
0.000290 main                        (2): libgphoto2 2.5.16
0.000296 main                        (2): libgphoto2 has been compiled with the following options:
0.000301 main                        (2):  + all camlibs
0.000305 main                        (2):  + /usr/bin/clang (C compiler used)
0.000309 main                        (2):  + ltdl (for portable loading of camlibs)
0.000313 main                        (2):  + EXIF (for special handling of EXIF files)
0.000318 main                        (2): libgphoto2_port 0.12.0
0.000323 main                        (2): libgphoto2_port has been compiled with the following options:
0.000327 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000332 main                        (2):  + /usr/bin/clang (C compiler used)
0.000336 main                        (2):  + ltdl (for portable loading of iolibs)
0.000340 main                        (2):  + USB (libusb1, for USB cameras)
0.000344 main                        (2):  + serial (for serial cameras)
0.000348 main                        (2):  + no resmgr (serial port access and locking)
0.000352 main                        (2):  + no ttylock (serial port locking)
0.000356 main                        (2):  + no lockdev (serial port locking)
0.000360 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000365 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000369 main                        (2): invoked with following arguments:
0.000374 main                        (2):   --debug
0.000378 main                        (2):   --debug-logfile=mylogfile.txt
0.000382 main                        (2):   --list-files
0.000420 load_settings               (2): Creating gphoto config directory ('/Users/phil/.gphoto')
0.000523 load_settings               (2): Loading settings from file '/Users/phil/.gphoto/settings'.
0.000659 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000676 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/opt/local/lib/libgphoto2_port/0.12.0'...
0.000833 foreach_func                (2): Called for filename '/opt/local/lib/libgphoto2_port/0.12.0/disk'.
0.001218 foreach_func                (2): Loaded '' ('^disk:') from '/opt/local/lib/libgphoto2_port/0.12.0/disk'.
0.001226 foreach_func                (2): Called for filename '/opt/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001500 foreach_func                (2): Loaded 'PTP/IP Connection' ('ptpip:') from '/opt/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001508 foreach_func                (2): Loaded '' ('^ptpip:') from '/opt/local/lib/libgphoto2_port/0.12.0/ptpip'.
0.001513 foreach_func                (2): Called for filename '/opt/local/lib/libgphoto2_port/0.12.0/serial'.
0.004855 foreach_func                (2): Loaded 'Serial Port Device' ('serial:') from '/opt/local/lib/libgphoto2_port/0.12.0/serial'.
0.004862 foreach_func                (2): Loaded '' ('^serial:') from '/opt/local/lib/libgphoto2_port/0.12.0/serial'.
0.004867 foreach_func                (2): Called for filename '/opt/local/lib/libgphoto2_port/0.12.0/usb1'.
0.043288 foreach_func                (2): Loaded '' ('^usb:') from '/opt/local/lib/libgphoto2_port/0.12.0/usb1'.
0.043309 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:020,011') from '/opt/local/lib/libgphoto2_port/0.12.0/usb1'.
0.043317 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:020,002') from '/opt/local/lib/libgphoto2_port/0.12.0/usb1'.
0.043325 foreach_func                (2): Called for filename '/opt/local/lib/libgphoto2_port/0.12.0/usbdiskdirect'.
0.043821 foreach_func                (2): Called for filename '/opt/local/lib/libgphoto2_port/0.12.0/usbscsi'.
0.044243 gp_port_info_list_count     (2): Counting entries (8 available)...
0.044256 gp_port_info_list_count     (2): 4 regular entries available.
0.044283 gp_abilities_list_load_dir  (2): Using ltdl to load camera libraries from '/opt/local/lib/libgphoto2/2.5.16'...
0.044498 foreach_func                (2): Found '/opt/local/lib/libgphoto2/2.5.16/adc65'.
[... cut ...]
0.095037 gp_filesystem_reset         (2): resetting filesystem
0.095053 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.095058 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.095062 delete_all_folders          (2): Internally deleting all folders from '/'...
0.095067 lookup_folder               (2): Lookup folder '/'...
0.095072 lookup_folder               (2): Found! / is 0x7f970f406a90
0.095079 recurse_delete_folder       (2): Recurse delete folder 0x7f970f406a90//

msmeissn, 2018-02-25:
i added a check in current git to fix this hopefully. will be in 2.5.17

closed 2018-02-25 by msmeissn

pwwong, 2018-02-25:
Thanks for your effort.

-phil