gphoto/libgphoto2 #235: Include Canon EOS M6

slepkaviba, 2018-02-09:
Hi,

I see that M6 is not in the list - is there any support plan for that model (I guess and hope its fairy similar to M5).

Br,
S.

msmeissn, 2018-02-10:
it probabluy will just work... do you have the USB ids? can you test?

slepkaviba, 2018-02-10:
usb:001,011

I can set up some Ubuntu VM and check if it helps.

msmeissn, 2018-02-10:
this is jus the bus location. can you run:

lsusb    

and quote the line of the camera?

slepkaviba, 2018-02-10:
God, was surprised it simply matches my ports...

ID 04a9:32c5

Here you go.

msmeissn, 2018-02-10:
If you have a gphoto2 installed, just try 
gphoto2 --capture-image
for starters :)

slepkaviba, 2018-02-10:
Ok, that one seemed to do the trick. Picture was taken.

msmeissn, 2018-02-11:
thx for the feedback, considering working similar to eos m5 / m10

closed 2018-02-11 by msmeissn