gphoto/libgphoto2 #238: Capture not supported by SONY UMC-R10C

sameenashaikh29, 2018-02-16:
Hi,
 I installed the gphoto2 library using the sudo apt-get install gphoto2 command. My camera was not getting auto detected. I then updated the library through https://github.com/gonzalo/gphoto2-updater/ and the camera got detetcted using the gphoto2 --auto-detect command. It is listed ined in the supported cameras list on gphoto2 on site and sudo --list-cameras too gave me this camera in the list. But capture-image and --trigger-capture command doesnt work.. Ive listed outputs of some commands below.

pi@raspberrypi:~ $ lsusb
Bus 001 Device 006: ID 413c:2107 Dell Computer Corp. 
Bus 001 Device 005: ID 0458:003a KYE Systems Corp. (Mouse Systems) NetScroll+ Mini Traveler / Genius NetScroll 120
Bus 001 Device 008: ID 054c:0a79 Sony Corp. 
Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pi@raspberrypi:~ $ gphoto2 --version
gphoto2 2.5.15

Copyright (c) 2000-2017 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.15         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.15         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

pi@raspberrypi:~ $ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Sony UMC-R10C                  usb:001,008     

pi@raspberrypi:~ $ gphoto2 --abilities
Abilities for camera             : Sony UMC-R10C                               
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Capture not supported by the driver
Configuration support            : no
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes

pi@raspberrypi:~ $ gphoto2 --summary
Camera summary:                                                                
Manufacturer: Sony Corporation
Model: UMC-R10C
  Version: 2.00
  Serial Number: 00000000000000004205006000000079
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: Association/Directory, MPEG, JPEG, Unknown(b982), Unknown(b301), Undefined Type, ARW, Unknown(b310)

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:

Device Property Summary:

pi@raspberrypi:~ $ gphoto2 --capture-image
                                                                               
*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.

Any h

msmeissn, 2018-02-20:
can you switch it to PC Control mode in the cameras USB menu?

sameenashaikh29, 2018-02-23:
It just has 2 modes. mass storage and PTP mode.. The camera gets detected in PTP mode

msmeissn, 2018-02-25:
This is weird, if that is using gphoto2 already, then it should work.
is the drotag device using USB protocol level triggering or some over the wire external trigger?

sameenashaikh29, 2018-02-28:
Hi,
     I am not sure about the protocal it is using..

DROTAG is a small board that simply sits between your camera and your Flight Controller (like Pixhawk),
allowing you to remote trigger the camera (manual or automatically) and have the taken pictures automatically
geo-tagged. Upon landing you only need to take out the microSD card from the DROTAG board
and the images are immediately available and ready to be visualized/used by your preferred stitching
software.
Currently just the main Sony Series cameras were tested (A5000, A5100 & A6000), although it is
likely that many more tethering capable cameras are supported. Our product is based on **libgphoto2** and
it’s camera compatibility list can be found in this link;

I've used the drotag with sony umc-r10 and it works just fine with the mode set to ptp.

sameenashaikh29, 2018-02-28:
Does it make a difference if it is not listed in the remote controlled camera's list??? Only file download is supported by the gphoto2 library for this camera. Can the other functions like trigger and capture image and download be added by you as it is much needed for my work???

sameenashaikh29, 2018-03-01:
Hi,sum
I tried using the file download feature as it is the only one listed in the device capabilities, but i tried commands for download but didn't succeed. I tried using the set capture target but to no avail. this parameter is not listed in the configs. All the parameters listed are read only.. 

pi@raspberrypi:~ $ gphoto2 --list-all-config
/main/actions/opcode                                                           
Label: PTP Opcode
Readonly: 0
Type: TEXT
Current: 0x1001,0xparam1,0xparam2
END
/main/status/serialnumber
Label: Serial Number
Readonly: 0
Type: TEXT
Current: 00000000000000004205006000000079
END
/main/status/manufacturer
Label: Camera Manufacturer
Readonly: 0
Type: TEXT
Current: Sony Corporation
END
/main/status/cameramodel
Label: Camera Model
Readonly: 0
Type: TEXT
Current: UMC-R10C
END
/main/status/deviceversion
Label: Device Version
Readonly: 0
Type: TEXT
Current: 2.00
END
/main/status/vendorextension
Label: Vendor Extension
Readonly: 0
Type: TEXT
Current: Sony PTP Extensions
END
pi@raspberrypi:~ $ gphoto2 --get-config capturetarget
*** Error (-2: 'Bad parameters') ***                                           

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-config capturetarget

Please make sure there is sufficient quoting around the arguments.

msmeissn, 2018-04-02:
I think the DROTAG uses external triggering using the flash shoe adapter. 

I fear your camera does not allow it over USB.

sameenashaikh29, 2018-04-05:
Ok if i have to do external triggering, its fine. I can do that. Why is the download not possible through gphoto2, or accessing the camera memory card not possible through gphoto2, although images download is a feature listed by gphoto2 for sony UMC-R10. Drotag does use it atleast for downloading images or accessing the memory card through gphoto2

cfort38, 2022-04-02:
Hi sameenashaikh29,
I just got a UMC-R10C: have you been finally able to drive this camera and how ? 
thank you