gphoto/libgphoto2 #239: EOS M10 "PTP Device Busy"

PhilippRaab, 2018-02-19:
Hi,

image capturing with EOS M10 is most of the time (except once) not working. The failure message on version 2.5.15.1:
```*** Error ***
storage 0xffffffff, but handle 0x00000000?

*** Error ***
Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.

root@booth:~# gphoto2 --version
gphoto2 2.5.15.1

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.15.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.16.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

msmeissn, 2018-02-25:
could you try current git of libgphoto2?

PhilippRaab, 2018-02-26:
With the current git it is working, thanks for the great support and your time!!
Is there some way that i can contribute to the project?

emanuelelaface, 2018-03-09:
I am experiencing the same issue. In particular it is triggered by any set config on the camera. If I use gphoto2 only to shoot it works well, but any time that I try to set iso, shutter speed or any other parameter it generates this problem.
I compiled gphoto2 and libgphoto2 from the latest git but the issue is still there.
I also did a small test to see which function create the issue and it is anything that sends the config back to the camera.
Both gp_camera_set_single_config or gp_camera_set_config hang the camera, I think there is something wrong in how the camera receive the configuration, but I am too ignorant of PTP to debug it.

msmeissn, 2018-03-12:
@emanuelelaface  can you also test the latest GIT of libgphoto2?

emanuelelaface, 2018-03-12:
@msmeissn I did it with the latest git of libgphoto2 on both linux (ARM64) and Mac.
Just as a test I tried to control my old CANON 1000D and everything works perfectly with that camera.

emanuelelaface, 2018-03-18:
I reinstalled everything from scratch compiled the libgphoto2 from the latest git, the latest gphoto2 but I still obtain PTP Device Busy when I try to set any parameter.

I attach the logfile generated by the command
env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config-index iso=4

[eos10m-logfile.txt](https://github.com/gphoto/libgphoto2/files/1822635/eos10m-logfile.txt)

brivbl, 2018-08-13:
Hello, 
I get the exact same problem with the latest version of gphoto2 and libgphoto2, have you found a solution to this problem yet ?

emanuelelaface, 2018-08-13:
Not really, I installed the CHDK firmware on the Canon and I am using that one with the chdkptp library to control the camera, and it works well.

brivbl, 2018-08-13:
I will try this then, thank you for your quick answer

argosvr, 2018-08-20:
well.. the EOS M10 is not usable with Gphoto2.
You can take pictures but no way with the last version of Gphoto2 to change the parameters of the camera.
CHDK used with Gphoto2 has the same problem: impossible to set the parameters.
The only solution is to not use Gphoto2 but directly the lib given by CHDK but the problem is that there are a lot of parameters you can't set.
I saw different issues on Github about Gphoto2 and the EOS M10 but they are still open without any answers.

emanuelelaface, 2018-08-20:
I am able to do almost everything (even more than the default feature like to shoot for more than 30 seconds) with CHDK.
If you want to use it maybe you can write me directly, this is not the right thread I guess.

emanuelelaface, 2018-09-02:
I am trying to use libgphoto2 through the wifi interface. When I try it immediately closes the communication and does not work, so I did some tcpdump and analyzed what the official app is sending to the camera. The sequence that i found is
- the application does an handshake where it sends to the camera its guid, name and version and gets back the camera name, a camera id and a connection id;
- with the connection id the application opens a second socket and sends to the camera that id, if this is not done within few seconds the camera closes the connection. This second socket is used to close the communication when the app is closed on the phone I think;
- on the main connection the app sends a PTP 0x1002 followed by a PTP 0x9114 to initialize the remote control;

From this moment on I think that the management of the camera is the usual PTP, even if some command (like the one for live view) probably is a bit strange, but I am not sure I haven't extensively debugged this.

It would be great to have the possibility to use the remote wifi control of the camera. The code that I used to do some test (in python) is the following.

`address='192.168.212.131'
port=15740
command_socket=socket.socket(socket.AF_INET, socket.SOCK_STREAM)
command_socket.connect((address,port))
packet_init=struct.pack("<I", 1).encode('hex') # 01000000
#guid='36791b9f2812405b8574430805ea1d8d'
guid='deadbeefdeadbeefdeadbeefdeadbeef'
name='6900500068006f006e0065000000' # i00P00h00o00n00e
version='00000100' # 00000100
packet_length=struct.pack("<I", len((packet_init+guid+name+version).decode('hex'))+4).encode('hex')
send_message=(packet_length+packet_init+guid+name+version).decode('hex')

command_socket.send(send_message)
recv_message=command_socket.recv(1518).encode('hex')

packet_length=struct.unpack("<I",recv_message[0:8].decode('hex'))[0]
packet_init=struct.unpack("<I",recv_message[8:16].decode('hex'))[0]
connection_number=struct.unpack("<I",recv_message[16:24].decode('hex'))[0]
guid=recv_message[24:56]
name=recv_message[56:112] # to be decoded?
version=struct.unpack("<I",recv_message[112:120].decode('hex'))[0] # In theory it should be 1 but it is not

control_socket=socket.socket(socket.AF_INET, socket.SOCK_STREAM)
control_socket.connect((address,port))
packet_init=struct.pack("<I",3).encode('hex')
packet_length=struct.pack("<I", len((packet_init+recv_message[16:24]).decode('hex'))+4).encode('hex')
send_message=(packet_length+packet_init+recv_message[16:24]).decode('hex')
control_socket.send(send_message)
recv_message=control_socket.recv(1518).encode('hex')

packet_init=struct.pack("<I", 6).encode('hex')
data_phase_info=struct.pack("<I", 1).encode('hex')
operation_code=struct.pack('<q', int('1002', base=16)).encode('hex')[0:4]
transaction_id=struct.pack("<I", tx_id).encode('hex')
session_id=struct.pack('<q', int('41', base=16)).encode('hex')[0:8]
packet_length=struct.pack("<I", len((packet_init+data_phase_info+operation_code+transaction_id+session_id).decode('hex'))+4).encode('hex')
send_message=(packet_length+packet_init+data_phase_info+operation_code+transaction_id+session_id).decode('hex')

command_socket.send(send_message)
recv_message=command_socket.recv(1514).encode('hex')

tx_id+=1
packet_init=struct.pack("<I", 6).encode('hex')
data_phase_info=struct.pack("<I", 1).encode('hex')
operation_code=struct.pack('<q', int('9114', base=16)).encode('hex')[0:4]
transaction_id=struct.pack("<I", tx_id).encode('hex')
session_id=struct.pack('<q', int('15', base=16)).encode('hex')[0:8]
packet_length=struct.pack("<I", len((packet_init+data_phase_info+operation_code+transaction_id+session_id).decode('hex'))+4).encode('hex')
send_message=(packet_length+packet_init+data_phase_info+operation_code+transaction_id+session_id).decode('hex')

command_socket.send(send_message)
recv_message=command_socket.recv(1514).encode('hex')`

msmeissn, 2018-09-03:
http://gphoto.org/doc/ptpip.php describes the protocol, the second one is the event pipe socket.

we seem to do this already same or similar as your python code.

msmeissn, 2018-09-03:
can you try with gphoto2 --port ptpip:IPOFCAMERA --camera "Canon EOS (WLAN)" -L  
or so?

apsylone, 2019-10-12:
Does someone found a solution ? I have the same problem on EOS 6D Mark II.

msmeissn, 2019-10-12:
to what? the device busy?

apsylone, 2019-10-12:
@msmeissn Yes, I have the exact same problem.

jamesarthur, 2022-06-08:
Hello. Came here after much googling. I have recently tried using gphoto2 to connect to my Canon M10. Here’s what I’ve found.
* I can connect to the M10 via a USB cable
* I can list images, capture images and transfer them via USB
* I cannot change any settings over USB. doing so seems to lock up the camera, requiring a cold boot
* None of the above works while the device is mounted or the gvfs-gphoto2 volume monitor process is running
* I can also connect via WiFi/ptpip. I choose the “connect to smartphone” option on the camera, then call gphoto2 with the —port ptpip:x.x.x.x option.
* The Wi-Fi connection terminates after one call to gphoto2 but I can use —shell to keep the connection open as long as I like.
* I *can* change settings over Wi-Fi: exposure, iso.
* I can also capture and transfer.

Hope that info helps someone.

jamesarthur, 2022-10-08:
Just experimenting with this a bit more, trying to see if I can get anything more out of my M10 over USB.

`$ gphoto2 --debug --debug-logfile=logfile.txt --set-config shutterspeed=5 --capture-image-and-download
                                                                               
*** Error ***              
Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

$ gphoto2 -v
gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.30.1       standard camlibs (SKIPPING lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking`

[logfile.txt](https://github.com/gphoto/libgphoto2/files/9740365/logfile.txt)

hkskoglund, 2023-04-21:
Hi! @jamesarthur mentioned that wifi connection terminates after a single gphoto2 command. Is this on the camera side,  so that the camera must be setup for wifi each time? I am planning to use M10 to capture timelapse over wifi with external power adapter . Is this possible over wifi?

jamesarthur, 2023-05-30:
@hkskoglund It’s been a while, but yes. The key will be keeping the connection open.

The camera will require you to initiate the Wi-Fi connection on its screen each time you want to connect; ideally you don’t want to do that each frame of your timelapse, so you need to make sure the connection doesn’t drop between commands. That’s ok though - shell mode does that.