gphoto/libgphoto2 #241: Gphoto2 Comms With Canon 5DS Fail Easily

J0hannB, 2018-02-25:
I have been happily using gphoto2 to communicate with a pair of Canon 6D's for some time now. I recently upgraded however to Canon 5DS', however, and have noticed that the gphoto2 communication with these cameras seems less robust. Specifically, making multiple subsequent gphoto2 calls without waiting a second or two the cameras switch into their "USB mode" or whatever it's called (i.e., screens show a white logo of a laptop, and the top LCD just flashes "busy" when any buttons are pressed). The same happens if I run a command while the camera is taking pictures (e.g., an external device is triggering the cameras repeatedly on an interval via the remote port and I call `--capture-tethered` while an image is being captured). 

I occasionally saw similar issues with the 6D's but this happens much, much more frequently on the 5DS'

J0hannB, 2018-06-18:
So far my solution has been to wait 4s between sending commands (I probably only need 2s, but I doubled it just to be safe). I also avoid sending any commands when the shutter is being triggered externally.

elstoc, 2018-06-21:
Symptoms (white logo of laptop and LCD flashing 'busy') seem the same as I'm getting on my 760D (https://github.com/gphoto/gphoto2/issues/131).

joelcomp1, 2019-01-29:
I think I am seeing this same issue after switching from the Mark III, did you call get the error: ERROR: Could not capture image. ERROR: Could not capture?   It seems to be like you described I saw very rarely on Mark III but now on 5DS I see much more often