gphoto/libgphoto2 #242: Changing autopoweroff Setting on Canon 5DS Not Working

J0hannB, 2018-02-25:
I noticed that `/main/settings/autopoweroff ` is returned when I call `--list-config` on my Canon 5DS, but whenever I call `--set-config autopoweroff=0` I get 
```
*** Error ***
The property 'Auto Power Off" / 0xd114 was not set (0x2019: PTP Device Busy)
```

I am able to set other properties, like `reviewtime` and exposure settings, just not `autopoweroff`

msmeissn, 2018-02-25:
can confirm this with a EOS 750D.

saiiro, 2018-04-06:
same problem with EOS 1300D, 1200D, 100D, 200D ...
200D, however, return a slightly different error (PTP I/O instead of Device Busy): 
`The property 'Auto Power Off' / 0xd114 was not set (0x02ff: PTP I/O Error).`

J0hannB, 2018-06-18:
I have not gotten it to work yet

samidalati, 2020-02-04:
I can confirm that this does not work on T7i, 80D, RP. So far I had to set auto power to off on the camera menu

msmeissn, 2020-02-09:
SDK documentation is also not very helpful, does not seem to mention it.

From libgphoto2 side, we are sending keepdeviceon commands to avoid powering it down.

eitijupaenoithoowohd, 2020-12-28:
Can confirm that issue with a EOS 600D.

How to use "keepdeviceon" with cli when I use
gphoto2 --stdout --capture-movie ?
I can't find similar option for gphoto2 --help

msmeissn, 2020-12-28:
we actually call it in the libgphoto2 preview capture, which is used during --capture-movie

                        if (ptp_operation_issupported(params, PTP_OC_CANON_EOS_KeepDeviceOn)) C_PTP (ptp_canon_eos_keepdeviceon (params));

so there is no need to do anything there. Is the camera powering off during --capture-movie?

which libgphoto2 version is in use?

eitijupaenoithoowohd, 2021-01-07:
> which libgphoto2 version is in use?
libgphoto2 2.5.16 (default in Ubuntu 18.04)

>Is the camera powering off during --capture-movie?
Yes it is. Within a range from one minute till three.
In my case the issue has been fixed with the last firmware update. I had have 1.0.2 before update and 1.0.3 now.
Now it works like a charm.

msmeissn, 2021-01-07:
I would say this was a camera side issue then -> closing.

thx for feedback!

msmeissn, 2021-01-07:
i can still reproduce autopoweroff not being able to set .. might not be possible from the pc

bwims, 2021-08-18:
I have the latest-1 version of the firmware for my 1100d.  Nothing in the release notes persuades me that the behaviour of the camera would change if I updated it.

My main need for autopoweroff is ensure that the camera stays on for a Gphoto2 session, but I forget to reset it, causing my battery to drain...

Possible workaround:
I have been looking at the EOS Utility manual for the 1100d.  

Under Preferences, there is a check box labelled "Auto power off (except when using an ac power supply)"

and the text says

> If you checkmark [Auto power off], the camera's auto power off 
function is enabled. If you remove the check mark, the auto 
power off function remains disabled, even if you set the auto 
power off function on the camera (only when the camera is 
connected to a computer.). 

This sounds to me like the EOS utility is doing the equivalent of PTP_OC_CANON_EOS_KeepDeviceOn

Question:  Is PTP_OC_CANON_EOS_KeepDeviceOn  something that can only last for the duration of an executable image, or can it be made persistent until turned off?

If it can only last for the duration of an image, would it be possible to create a utility that calls it, then prompts for CLI commands that it executes, waiting for a command to terminate the image?

Thanks!

B.