gphoto/libgphoto2 #243: Add EOS M2 support

anagnost, 2018-03-03:
Already posted this to the mailing list, but let it be here just for the case.

At the link below I have posted my patch for EOS M2 support. Key features of this camera, which make it different from other of its series: 1) it is necessary to reget device info after setting remote mode to reveal hidden capabilities; 2) it uses a "normal" EOS capture/AF method.

[https://gist.github.com/anagnost/c786fb215bef1771e268edbbd2204445](https://gist.github.com/anagnost/c786fb215bef1771e268edbbd2204445)

Blueshawk, 2018-03-06:
Any hope for my poor old Eos-m(1) from this?

anagnost, 2018-03-06:
The key difference between M(1) and M2 is that the later supports WiFi remote shooting. Since Canon implements this via the PTP/IP protocol, they could not completely remove the code responsible also
for the USB capture: just masked it a bit. On M(1) it may well be really crimped.

That being said, I would still be interested to have a look at the logs of your camera, if you can test it with libgphoto2 from the current git (as Marcus has already partially applied my patch)...

Blueshawk, 2018-03-07:
Been busy in the aftermath of a horrid windstorm. I'll get them asap.

Blueshawk, 2018-03-10:
I hit a snag after updating and getting things to compile again(been a while) The current code apparently has the m(1) locked out, or am I not holding my head right? 
[eos-m-log.txt](https://github.com/gphoto/libgphoto2/files/1799761/eos-m-log.txt)

anagnost, 2018-03-10:
Hmmm... Are you sure you really have compiled the current trunk? The log show your libgphoto2 version is 2.5.14.1, while it should be 2.5.16.1.

Anyway, the code doesn't lock anything, it just checks if specific PTP operations are supported. And that's why it is important to use the current version: it attempts to reget the device info (including the list of supported operations) at a specific point where they may have changed. This is crucial for M2, so I was interested to check if this changes anything for M.

Blueshawk, 2018-03-10:
o ....I bet the branch is still set to something obtuse from when I was helping with pentax. Thank's I'll sort that out.

msmeissn, 2018-03-12:
It does not seem to offer the remote release functionality nor the getevent log , which makes remote capture unlikely

anagnost, 2018-03-12:
Yes, I see: according to the log, no changes in DeviceInfo after setting remote mode, so most probably nothing more to do. It was interesting to ensure that though.

Blueshawk, 2018-03-12:
Worth a try. Thanks for looking guys.  :)

msmeissn, 2018-03-30:
general issue is now done I think

closed 2018-03-30 by msmeissn