gphoto/libgphoto2 #245: Canon 80D PTP General Error

w0jtek, 2018-03-12:
Hi,
I'm getting "PTP General Error" when trying to capture using this command:
gphoto2 --capture-image-and-download
My camera is Canon 80D and I've set full raw + jpeg to be saved. This has someting to do with that mode, because for example mraw + jpeg works fine.
When the issue occurrs in gphoto, I see "Error 70" on the camera's screen. Then I have to extract the battery, turn the camera off and on again... Any idea what might be wrong?

msmeissn, 2018-03-12:
https://sourceforge.net/p/gphoto/feature-requests/479/   has the logfile

Can you please update to libgphoto2 2.5.16 or current GIT for testing too?

msmeissn, 2018-04-13:
I think I fixed this bug in current git (for 2.5.17 release)

msmeissn, 2018-04-17:
is liokely fixed in 2.5.17. if not please comment/open new issue

closed 2018-04-17 by msmeissn