gphoto/libgphoto2 #246: K-3II not working in Bulb mode

mccodesurfer, 2018-03-14:
I'm using Kstars/StellarMate Astro software which uses libgphoto2 for Pentax K-5II control and its working but when trying to do the same thing with a K-3II, the shutter opens and closes immediately. I tried the K3 in both bulb modes - i.e. click and hold; and click to start, click to stop - and tried it in MSC and PTP mode. I could not get a log file of any significance, and I could not install gphoto2 to get a log file that way. Hope you have some ideas. thanks

knro, 2019-01-14:
It seems this problem exists using the latest gphoto as well... is there any information we can submit to help diagnose this behavior for this specific model?

mccodesurfer, 2019-01-17:
I have a K-3 and linux box to work on but I not that versed in linux development so I would need some support or at least confirmation that @msmeissn be involved. (Unfortunately the K-3 and K-5 are not supported by Pentax's new SDK so no help there.)

msmeissn, 2019-01-18:
I do not have much knowledge of the Pentax series. I have one that works with Bulb, but not much knowledge about the others.
We sync / import the pentax code from the pktriggercord project, perhaps you could contact them.

mccodesurfer, 2019-01-18:
OK, I'll check pktriggercord. Thanks.
(I should clarify that K-5 is working on gphoto2 but as you say, limited with bulb mode, which is fine for astrophotography.)

mccodesurfer, 2019-01-19:
I’ve checked out pktriggercord and built it successfully.  I’m a little familiar because I was trying to drive the k-5 on macOS. Ill see what I can do but just stumbling through it right now.

mccodesurfer, 2019-01-20:
I tried the latest gphoto2 and got this:

pi@raspberrypi:~ $ gphoto2 --capture-image --bulb=5                        
Bulb mode enabled (exposure time: 5s).                                         

*** Error ***              
This camera does not provide any configuration options.
ERROR: Could not set bulb capture, result -6.
*** Error (-6: 'Unsupported operation') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image --bulb=5

Please make sure there is sufficient quoting around the arguments.

msmeissn, 2019-01-20:
I checked and pktriggercord has implemented a "new bulb" method. I need to port that also to libgphoto2

mccodesurfer, 2019-01-20:
I have info from asalamon at pktriggercord and he says the k-3II fix is not yet released but is will be soon.

mccodesurfer, 2019-01-20:
That would be great. Thanks!

mccodesurfer, 2019-02-07:
Can you tell me how pktriggercord is used? As a library? Do you issue command line type calls to it from gpotolib2? thanks

msmeissn, 2019-02-07:
i copy over parts of the code into camlibs/pentax/ and there is glue code tying it to libgphoto2 internals.

mccodesurfer, 2019-02-07:
Because gphoto2 and PkTetherCord doesn't work on the Mac for Pentax. Its the way the USB is handled in pktethercord. However it does work for Canon. Can you direct me to the USB handler for that? I wonder if I can copy the code to pktethercord.