gphoto/libgphoto2 #248: EOS-M: can't work with larger files

janbruns, 2018-03-16:
I have problems downloading larger files from an EOS-M.

Limiting USB-Port to low (full) speed, and upgrading to libgphoto2 2.5.16 didn't help.

It feels like there'd be some secret file-size limit (for cr2-files probably around 28MB) that decides wether the download would normally work, or never. And from the gphoto2 logs I see strong correlation with some "0x20" canon flag (which gphoto source code seems to interpret as "new"-indicator). May this flag probably suggest choosing another command for file transfer?

[try_all.txt](https://github.com/gphoto/libgphoto2/files/1820682/try_all.txt)
[try_IMG_2892.CR2.txt](https://github.com/gphoto/libgphoto2/files/1820683/try_IMG_2892.CR2.txt)
[try_IMG_2909.CR2.txt](https://github.com/gphoto/libgphoto2/files/1820684/try_IMG_2909.CR2.txt)
[try_IMG_2911.CR2.txt](https://github.com/gphoto/libgphoto2/files/1820685/try_IMG_2911.CR2.txt)

janbruns, 2018-03-16:
There's something more that points me to this idea:

The files that don't work also don't make the camera LED blink when the transfer is initiated. The camera seems to immediately decide the file doesn't work as soon as it's requested, probably similar to non-existing objects.

msmeissn, 2018-03-17:
0.334989 ptp_usb_sendreq             (2): Sending PTP_OC 0x1009 (Get object) (0x9190b5d1) request...
0.336157 ptp_usb_getdata             (2): Reading PTP_OC 0x1009 (Get object) data...
24.051317 ptp_usb_getresp             (2): Reading PTP_OC 0x1009 (Get object) response...

can you get 1 debuglog with data inside perhaps? most interesting is the Get Object response code, if its 0x2001? 

Lets hope we do not need from reegualr GetObject to one of the PTP_OC_CANON_EOS_GetPartialObjectEx  flavors :/

janbruns, 2018-03-17:
Looks like the camera locks down on GetObject and doesn't even respond.
[try_IMG_2911_with_data.CR2.txt](https://github.com/gphoto/libgphoto2/files/1821493/try_IMG_2911_with_data.CR2.txt)

janbruns, 2018-03-17:
Just tried out what happens if I move the /* clear the "new" flag on Canons */ code in library.c some lines up:

The file-flag gets cleared on the camera, but GetObject still fails.

janbruns, 2018-03-17:
Seems like getpartialobject fails...  modified library.c, like this.
On a virtal Box with winXP and no canon drivers, the problematic files also can'tt be downloaded. Currently trying to install the canon software on that box. The camera can work with that file.

		size=ob->oi.ObjectCompressedSize;
		if (size) {

                        uint32_t xlen;
                        unsigned char 	*ximage = NULL;
                        C_PTP_REP (ptp_getpartialobject (params,
                                oid, 0, size, &ximage, &xlen));
                        CR (gp_file_set_data_and_size (file, (char*)ximage, xlen));
//                        break;
#if 0
			uint16_t	ret;
			PTPDataHandler	handler;

			ptp_init_camerafile_handler (&handler, file);
			ret = ptp_getobject_to_handler(params, oid, &handler);
			ptp_exit_camerafile_handler (&handler);
			if (ret == PTP_ERROR_CANCEL)
				return GP_ERROR_CANCEL;
			C_PTP_REP (ret);
#endif
		} else {

[try_IMG_2911_with_data6.CR2.txt](https://github.com/gphoto/libgphoto2/files/1821622/try_IMG_2911_with_data6.CR2.txt)

janbruns, 2018-03-17:
EOS Utility had no probs with the file.

Here's a wireshark usb dump of connecting the camera to the virtual XP box, and using EOS-Utility to download one of the the problematic files: abnuto.de/jan/ws_canon_xp.pcapng (about 30MB)

msmeissn, 2018-03-17:
Can you try this patch. It does partial reading in blobs of 1MB, same as your USB trace.

[eos.txt](https://github.com/gphoto/libgphoto2/files/1821822/eos.txt)

janbruns, 2018-03-17:
The patch seems to work. It make gphoto download the problematic files. But the progress-indicaton is broken now.

msmeissn, 2018-03-17:
I have to think on how to handle the progress here, to avoid displaying multiple progress reports :/

janbruns, 2018-03-17:
Great. What an ugly prob this was ("man, use a bettern cable" all the time).

closed 2018-03-28 by janbruns

msmeissn, 2018-03-28:
just to cross check, was it a bad cable?

janbruns, 2018-03-29:
Hmn, I of course tried multiple cables. This is one of the first tings I try when I have poblems with USB connections (just after trying out another port). 

The wireshark snif ( abnuto.de/jan/ws_canon_xp.pcapng ) with the virtual winXP box was recorded using a relatively long, thin cable, that now (after your patch) works well with 480 bps, but for that snif, I downgraded the usb port to 12 bps by typing (into a root console)

echo 2 > /sys/bus/usb/devices/usb1/../companion

where 1 and 2 are numbers from "lsusb -t". Not my idea, don't ask me how this works, but it works this way for me (somehow, the above file isn't the same as /sys/bus/usb/devices/companion as one could expect).

janbruns, 2018-03-30:
Just in case there's something unexpected in the wireshark log: it was interleaved by some other usb hid device communication on the same root hub. It was possible to filter that out (and I did that before uploading the file, since stupid me moving the mouse didn't seem to be interesting enough).

For example, wireshark interprets 3 packets as interupt_in on Endpoint 3. Not sure if this interpretation is reliable. The file I downloaded from the camera (that theoretically should be completely reconstructable from the log) was a 29MByte CR2, so 31MB log didn't add much overhead (and the hid events I removed didn't add much file-size).

msmeissn, 2018-03-30:
I just wanted to cross check your comment ;)

that confirms 
- cables not an issue.
- patch fixes the problem.

thx :)

janbruns, 2018-03-30:
I see. It's a little suprising that noone complained before, given the eos-m might be relatively popular.
But given the problem only arose for large, noisy raws and videos, but never for jpegs and normal raws (here, for me), the remaining people that might have had this problem probably silently just decided to use external card readers instead.

msmeissn, 2018-03-30:
Yeah, I was wondering about this too. Perhaps they only wondered and did not report a bug ;)