gphoto/libgphoto2 #251: Windows Port - CAMLIBS Default Path Issue

Oxydation, 2018-03-25:
I am currently working on a project to provide `libgphoto2 `on multiple platforms via C#. To achieve this I am working with [SharpCamera](https://gitlab.com/define-private-public/SharpCamera) from @define-private-public. Despite some startup problems I got it running from C# code on a Win 64-bit machine with 32-bit dlls of `libgphoto2 `(check this [issue](https://gitlab.com/define-private-public/SharpCamera/issues/10)). 

Future goal is to ship a NuGet package with all dependencies. We want to make sure, that there is less or no setup hassle for users. I succeeded in putting all dependencies (`libgphoto2-6.dll, libgphoto2_port-12.dll, libexif-12.dll, libintl-8.dll, libsystre-0.dll, libgcc_s_dw-1.dll`) as well as the IOLIBS (`usb1.dll`, `disk.dll` and `ptpip.dll`) right next to the application. I am using those libraries provided by the MYSYS2, MINGW32 environment.

Now to my problem: I could not load the camlibs if I put them next to the` libgphoto2-6.dll` or in a `CAMLIBS `directory in the same directory of `libgphoto2-6.dll`. Please check the given [issue](https://gitlab.com/define-private-public/SharpCamera/issues/10#note_64959496) for more information. 

It seems that during the build process a not existing path has been added to the libgphoto2-6.dll, where instead should be just a dot `.`. It represents the value of the `CAMLIBS ` define. As you can see in the issue, I had to modify the DLL via hex editor to change the wrong path to the` .` (dot). After this modification I can run the application with all cam-libs relative to the `libgphoto2-6.dll`.
[image]

Some - maybe important - detail: I can set the environment variable in Windows (CAMLIBS, `.`) then it was also working. But if the environment variable was not set, it used the weird path from the DLL.

Oxydation, 2018-10-31:
Yes, I was able to use SharpCamera on Windows, as stated in the issue and the linked issue of the SharpCamera-Repo.

define-private-public, 2018-11-01:
@Oxydation What are the changes you've made?  I'd really like to get SharpCamera to work properly on Windows.

Oxydation, 2018-11-02:
All changes done are in my fork of SharpCamera under the branch **wintest**: https://gitlab.com/Oxydation/SharpCamera/tree/wintest

You can find a documentation on how to get it running on Windows (docs/SharpCamera_on_Windows.pdf). I hope everything is there - ping me if something is missing or not working.

- Hint 1: I did not merge with your repository yet (.net standard etc.).
- Hint 2: With these changes made it might not work on the other platforms anymore (dlls etc.). That's why it is on the wintest branch which is for testing purposes. I added pre-built libgphoto2 libraries for Windows to the project to ease the build process.

mms-, 2019-04-12:
@Oxydation I am trying your wintest branch, and running the SimpleCapture app under x86 like you say I get 'Unable to load DLL 'libgphoto2_port-12': The specified module could not be found. (Exception from HRESULT: 0x8007007E' error.

All 74 dlls from your libgphoto2 project, including libgphoto2_port-12.dll are in the same folder as SimpleCapture.exe. Any other dlls needed?

I didn't do Zadig driver replacement yet, I don't have a device connected.

Any ideas?