gphoto/libgphoto2 #255: Canon EOS 200D Raw capture fail

saiiro, 2018-04-04:
We've been using libgphoto2 for a long time to capture RAW (.cr2) files from various canon cameras and it works like a charm. 
Now we've recently bought new SL2 (EOS 200D) and our program crashes when trying to capture *some* cr2. 
Crash can be reproduced (although not systematically) using RAW settings on camera, using the following commandline and pressing the trigger. 
`gphoto2  --wait-event-and-download`

Now the interresting thing is that it works ok on completely dark images with a filesize < 30 Mb and seems to crash on more complex images that have a filesize > 30 Mb (up to 36 Mb). The errors seems to point toward: 
`PTP_OC_CANON_EOS_GetPartialObject.`

My 2 cents is that there is a int32 overflow somewhere but I cant say for sure... 
Debug  log is attached. 

Thanks ! 

[raw_200D.txt](https://github.com/gphoto/libgphoto2/files/1877966/raw_200D.txt)

janbruns, 2018-04-05:
Pretty much exactly sounds like #248 I reportet some week ago, and that was solved by Marcus Meissner's patch (in that case, introducing ptp_canon_eos_getpartialobject**ex** and using i in place of simply getObject was the solution). 

Not sure about if this change already made it's way to direct captures (my camera can't do that anyway, just like with canon eos-utility), but maybe it's worth trying to update.

msmeissn, 2018-04-05:
yes, can you please test current libgphoto2 git ...

saiiro, 2018-04-05:
Yeah the symptoms are exactly the same as #248. I'll try to test the current git and report when its done. Thanks guys !

saiiro, 2018-04-05:
Ok, I tested with the git version and it does fix the problem ! 
Thanks a lot @msmeissn and @janbruns , closing the issue now.

closed 2018-04-05 by saiiro