gphoto/libgphoto2 #265: libgphoto2 v2.5.17 causes corrupted image transfers from Canon EOS camera(s)

ccrtnsp, 2018-04-28:
After upgrading libgphoto2 to latest Debian packaged version 2.5.17, I immediately noticed that the CR2 RAW files fetched via "gphoto2 -P" were unreadable by Darktable, Adobe Lightroom and other RAW image software. My cameras are Canon EOS 7D Mark II and Canon EOS 500D, connected via USB. The problem occurs with both cameras.

I quickly noticed that repeated transfers often resulted in files with different SHA256 sums. Sometimes the transfer did succeed, but this seems very random. Downgrading to libgphoto2 to previous 2.5.16 makes the problem go away, checksums are always consistent. With 2.5.17, each file had different checksum for each transfer, pointing to serious corruption.

I also tested with latest GIT versions  of libgphoto2 (91a8425a4fa27def793fa9db2bcb4a71c26c927b) and gphoto2 (5e74760beabf3c57b7c6b3fc2ee1bacf26ed9b07), the problem exists with those as well.

I can provide gphoto debug logs of transfer of one photo with both working and bugged versions at request, but those are rather large (about 100M each uncompressed, 75M xz'd).

Originally reported this to Debian bugtracker, https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=897059

ccrtnsp, 2018-04-30:
Bisected the problem to commit def1af78b1f62e7e099ec6c7ad7b0d54ddd0483c "Add partialobjectex reading for EOS". Reverting this on top of head fixes the issue.

msmeissn, 2018-04-30:
i would have had the suspicion, but the code first looks okish... hmm

msmeissn, 2018-04-30:
wjhat you could do, without having the data... if you download the same file 2 times, 

does the size differ?
if you run "cmp try1 try2"  wjhere is the offset of change?

msmeissn, 2018-04-30:
(my EOS cameras are packed for a move, I cannot test this before end of this week_)

ccrtnsp, 2018-04-30:
Size does not differ, but according to cmp the first different byte seems to always be the same (I did not do much testing yet, just few times, so not 100% certain):
$ cmp 1/IMG_3608.CR2 6/IMG_3608.CR2 
1/IMG_3608.CR2 6/IMG_3608.CR2 differ: byte 6291457, line 37282

ccrtnsp, 2018-04-30:
I have to note that _sometimes_ the transfers are correct, and _sometimes_ I can get several transfers in row correctly, but usually if I transfer 6 times, at least one of them will produce incorrect file.

msmeissn, 2018-04-30:
this is 0x600001  ... some thing with the partialobjectex transfer is not working, either the function is unstable, or I am passign incorrect parameters

ccrtnsp, 2018-04-30:
I seem to have stated a falsehood in my original report - after testing dozen times, I can't reproduce the bug on my EOS 500D, only on EOS 7Dmk2. So perhaps it is an problem with 7D2's PTP .. ?

heelstraf, 2018-05-01:
I can say that i have also this problem with 2.5.17-1.1 (openSUSE Tumbleweed)
Camera is EOS 6D, Darktable makes the information "failed to read camera white balance information from *.cr2"

the version 2.5.16-1.2 is fine, the rpm package comes from th openSUSE Tumbleweed repo.

matclab, 2018-05-12:
Same problem on libgphoto2-2.5.17-1 on arch linux with EOS 7D, with gphoto2 -P. Some photo are only partially downladed.

Reverting to libgphoto2 2.5.15 solve the problem

msmeissn, 2018-05-13:
I fixed this in 2.5.18, with commit a256a3fcccdd3719b629996c9651a849d8bb16ec

closed 2018-05-13 by msmeissn

heelstraf, 2018-05-17:
I confirm that 2.5.18 fixed this bug.
thx for the work!